InsanityAutomation/Marlin #15: Creality_DWINTest - LCD Volume setting does not save

camprevail, 2018-12-23:
For now I will just disconnect the speaker, lol

InsanityAutomation, 2018-12-25:
Just to put the answer discussed elsewhere here as well, this is partly by design. The original code did unsafe eeprom writes on every volume adjustment. It was set to default language to English and volume to 50% in order to eliminate the constant and unsafe eeprom actions. Potentially as I look at merging some of this into the marlin 2.0 extui API, volume storage might come back but for the time being there isn't a simple safe solution to storing the last setpoint. You can however modify the default value when you flash to make it always start wherever you want!

InsanityAutomation, 2019-05-16:
Note - Marlin 2.0 has had a wrapper for ExtUI eeprom settings so this may be feasible as I continue on that port.

LMS0815, 2019-08-29:
Just the opposite question...
Is there any GCODE to trigger an alert signal after print is ready?
Would be nice to play the "start tune" on demand...
Could not find any...

InsanityAutomation, 2019-08-29:
I can tie it to the buzzer at some point most likely. I might chase into it while im doing the rest of the eeprom stuff.

LMS0815, 2019-08-29:
THX:
` M117 BUSSER `

LMS0815, 2019-08-30:
Shouldn't M300 disable volume?
http://marlinfw.org/docs/configuration/configuration.html#speaker

> The duration and frequency for the UI feedback sound. Set these to 0 to disable audio feedback in the LCD menus. Test audio output with the G-code M300 S<frequency Hz> P<duration ms>

InsanityAutomation, 2021-06-20:
Done, in next release.

closed 2021-06-20 by InsanityAutomation