InsanityAutomation/Marlin #19: Creality_DWIN baby stepping limitation

trouch, 2019-02-06:
Hi,

Using `Creality_DWINTest` branch on a CR-10S Pro with an updated screen as well, I'm not able to setup a Z-Offset bellow `-0.53` using the live adjust / babystepping feature.
The value changes to `-0.54` when tapping the minus button but then quickly goes back to `-0.53`.

When setting up a Z-Offset below or equal to -0.6 in the levelling setting screen, I can see the proper value in the live adjust one. But as soon as I hit either minus or plus button here, it changes to `-0.53` 😬 

I've looked into the `Creality_DWIN.cpp` source code, mainly in the `case Zoffset` part, but I don't see any mistake there, and I tested the conversion code aside. I wonder if it could be a limitation from the screen itself since `Z_PROBE_OFFSET_RANGE_ = [-20;20]` 

I'll do more extended debugging with a UART probe, and submit a PR eventually but I'd like to get your advice first.

InsanityAutomation, 2019-02-18:
Unfortunately the screen itself stops sending commands at that point. Without the screen source, theres nothing more to do in the firmware side.

jbcpollak, 2019-04-03:
I've seen a similar issue but the values were something like -1.04 and -1.05, so incase anyone else stumbles on this, it is not the actual values that are the problem.

The only work around I've found is to reboot the printer.

InsanityAutomation, 2019-04-03:
I've been playing with it as I'm working on the 2.0 port as well, so not forgotten!

mellow129, 2019-05-15:
Thank you for your work!
I'm tracking this as I have the same issue. 0.5mm doesn't leave much adjustment room, and with the unreliability of the sensor I would really appreciate the ability to have more than 0.5mm of adjustment. I've already had the nozzle crash and melt through the top layer of the build plate because of that damn sensor.

InsanityAutomation, 2019-06-23:
Found the issue was a race condition between the lcd and firmware. Creality had the firmware send back the current z offset after processing which overwrote the incrementer in the lcd. This seemed to catch it in a loop. Ive modified the code to only force the value back when modified outside the adjust screen or when the adjustment has been clamped by min/max limits.

closed 2019-06-23 by InsanityAutomation

trouch, 2019-06-23:
Good catch ! Thanks.

jbcpollak, 2019-06-23:
To get the update do we just take the creality branch?

InsanityAutomation, 2019-06-23:
DwinTest branch is the 1.1.9 base