InsanityAutomation/Marlin #21: Cannot connect to printer after firmware update. Baudrate 115200 not working. 250000 does work

Ericvf, 2019-02-13:
Hi.

After updating to the firmware in this repo I am no longer able to connect to the printer over Serial using baud rate 115200. Luckily I can still re-flash and I have re-flashed to make sure it was not a one-off issue. I've also compiled the sources instead of using the included binary (hex file). 

Cura no longer works for me and does not detect the printer. I cannot update the firmware using Cura to original. 

Simplify3d works only with baud rate 250000.

What could cause this and is there a solution?

profpylons, 2019-02-13:
In [confiiguration.h](https://github.com/InsanityAutomation/Marlin/blob/a1a8d19a9d3d2587ff582e18e9c4d16aa451ec25/Marlin/Configuration.h#L119) (line 119, currently) you can change the baud rate before compiling if you need to. Valid values are shown above the definition.

Ericvf, 2019-02-13:
Thanks Paul, 

That really eases my mind since I now know this is intended. Can you elaborate on why this decision was made? What is the reasoning behind it? Please note that Cura's baud rate detection fails on my printer so it might be a good idea to disable this by default.

Again thanks

InsanityAutomation, 2019-02-18:
250000 is the stock on the machine from creality and the default for most printers. It was chosen due to the relationship to the clock speed on many early printer boards as jitter was reduced.

closed 2019-05-16 by InsanityAutomation