InsanityAutomation/Marlin #22: CR-10S Pro, ABL data different after restart machine

bangity, 2019-02-27:
### Description

### Steps to Reproduce

1. Go to Leveling and click Measuring
2. Turn printer off and on
3. Leveling data will change

**Expected behavior:** ABL data should remain the same (written to EEPROM)

**Actual behavior:** ABL data changed/Not Saved

[Configuration.zip](https://github.com/InsanityAutomation/Marlin/files/2910582/Configuration.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-03-11:
I have seen the reversed display and im working to fix it in the 2.0 port. The data is correct in eeprom, just the display order is backwards.

chunter1, 2019-05-14:
> I have seen the reversed display and im working to fix it in the 2.0 port. The data is correct in eeprom, just the display order is backwards.

With "2.0 port" you mean the "CrealityDwin_2.0" branch?
I tried to compile that one with Arduino 1.8.9 but got these errors:

`WARNING: Category 'LED' in library CHT16K33 LED Matrix Library is not valid. Setting to 'Uncategorized'
WARNING: Category 'Input' in library MechInputs is not valid. Setting to 'Uncategorized'
WARNING: Category 'Input' in library SerialSensors is not valid. Setting to 'Uncategorized'
C:\Users\campeg\AppData\Local\Temp\ccwahpTK.ltrans3.ltrans.o: In function `main':

<artificial>:(.text.startup+0x648): undefined reference to `ExtUI::leveling_is_valid()'

<artificial>:(.text.startup+0x652): undefined reference to `ExtUI::z_values'

<artificial>:(.text.startup+0x654): undefined reference to `ExtUI::z_values'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.`

InsanityAutomation, 2019-05-16:
The issue mentioned was just fixed @chunter1 but there are still quite a few things not implemented. Ive been extending ExtUI upstream to support everything needed as I work on it.

InsanityAutomation, 2019-06-23:
Fixed for B7

closed 2019-06-23 by InsanityAutomation