InsanityAutomation/Marlin #24: Enhancement Request - M117 Messages to LCD

mwolter805, 2019-03-05:
Hello,
Again, thank you for your work on this project! Requesting M117 messages be passed to the LCD. It would be helpful if M117 messages passed in (from OctoPrint for example) are able to be displayed on the LCD screen. There is quite a bit of helpful info that can be presented, such as the OctoPrint IP address, remaining print time, etc.

mwolter805, 2019-03-05:
silent_ninja1 answered this question on Reddit.

Unfortunately not as there is no always present data field to write it to... Nothing that takes a string either. Just numerical status values.

InsanityAutomation, 2019-03-11:
As a side note, thats me on Reddit as well :)

InsanityAutomation, 2019-05-16:
Closing as this just isnt possible with the creality screen files, as disappointing as that is.

closed 2019-05-16 by InsanityAutomation