InsanityAutomation/Marlin #25: Simplify3d issues after flashing

fr33dimension, 2019-03-11:
Since I flashed the tiny machine firmware for the CR10S Pro, the printer stops during usb printing from Simplify only. I Repetier no problem, but in S3d it stops. Sometimes with pause resume I can get it to continue for a few seconds and then the same thing happens. The gcode if loaded onto an sd runs fine.
Has to be something with ports or so any ideas? I am on 250000 baud, tried 11500 no change.

InsanityAutomation, 2019-03-11:
Usually the cable is to blame in these situations, however S3D can also be a bit unreliable. It crashes on G29 T topology reports for example. It seems that its parser has trouble handling certain data. I dont recommend printing directly from S3D as there is no guarantee it will be stable as host protocols evolve moving into Marlin 2.0

fr33dimension, 2019-03-11:
Thanks, interestingly though, with the CR firmware, that didn't happen.

InsanityAutomation, 2019-03-11:
> Thanks, interestingly though, with the CR firmware, that didn't happen.

If its not the cable, its most likely having issues parsing one of the added debug outputs and crashing. Question is what exactly. If you are able to capture a serial log of it, it will help when we get down to this stuff in the Marlin 2.0 port!

fr33dimension, 2019-03-13:
I was able to capture this will this help?
READ: X:77.00 Y:50.00 Z:4.95 E:-19.78 Count X:6160 Y:4000 Z:2000
READ: ok P15 B0
SENT: M105
SENT: G92 E0 
READ: ok P14 B1
READ: ok P0 B0
SENT: M105
READ: ok P0 B0
SENT: G1 Y80 E30 F15000 
READ: ok T:220.00 /220.00 B:60.01 /60.00 @:60 B@:23
SENT: G1 Y100 F5000 
SENT: T0
READ: X:5.00 Y:10.00 Z:0.20 E:0.00 Count X:5076 Y:3398 Z:2024
READ: ok P0 B0
SENT: G92 E0.0000
READ: ok T:220.04 /220.00 B:60.01 /60.00 @:59 B@:23
READ: ok P1 B0
SENT: M105
READ: ok P0 B0
SENT: G1 E-5.0000 F3600
READ: echo:Active Extruder: 0
Active Extruder: 0
READ: ok P15 B0
SENT: M105
SENT: G1 Z0.250 F1002
READ: X:5.00 Y:100.00 Z:0.20 E:0.00 Count X:400 Y:8000 Z:76
READ: ok P15 B1
READ: ok T:220.00 /220.00 B:60.01 /60.00 @:59 B@:20
READ: ok P14 B3
READ: wait
READ: ok T:220.00 /220.00 B:60.01 /60.00 @:59 B@:20
SENT: G1 X141.231 Y167.075 F6000
READ: ok P13 B3
SENT: G1 E0.2000 F3600
READ: ok P0 B2
SENT: G92 E0.0000
READ: ok P0 B2
SENT: M105
READ: X:141.23 Y:167.08 Z:0.25 E:0.00 Count X:6589 Y:11047 Z:89
READ: ok P0 B2
SENT: G1 X130.769 Y167.075 E0.6198 F1350
READ: ok T:220.00 /220.00 B:60.00 /60.00 @:59 B@:24
READ: ok P0 B3
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait

InsanityAutomation, 2019-05-16:
By the look of this the machine is waiting for more commands and S3D simply stopped sending them.

InsanityAutomation, 2019-08-29:
Since this appears to be an issue with S3D, and probably not relevant at all with 2.0, im closing this one out.

closed 2019-08-29 by InsanityAutomation