InsanityAutomation/Marlin #26: CR-10S PRO closes printing window after started print

Sciuruzs, 2019-03-19:
Hi

Just as the printer starts to print the printing window with al the adjustments closes and throws me back to the main screen. The print continiues but there is no way to get to the setings anymore.
Is there a fix for this? Using the newest version and the standard blue theme

Ericvf, 2019-03-28:
This happens to me also when printing over USB. Are you printing over USB or are you using the SD card? I've switched back to using the SD card for now. I'm one of those people who always prints using USB. I've never had an issue with it and it saves so much time compared to putting it on the SD card and constantly managing the machine using it's interface.

todddye, 2019-03-29:
I print only using the SD card and have this same issue. I did find that if you select the same file to print again it will allow you to go to the adjust screen.

Sciuruzs, 2019-03-29:
I found out the solution. The Menu is triggered ny a code in the start script.
Use this code. That worked for me 

G28 ; home
M420 S Z5 ; call on the auto level gods
G1 X20 Y25 F1500 ; avoid binder clips
G1 Z0.4 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 Y120 E10 F600 ; prime nozzle
G1 Y150 F5000 ; quick wipe

todddye, 2019-03-29:
Thats great.

Which code in the above calls the print menu?

todddye, 2019-03-29:
I found it. I commented out my M84 E and it works correctly now.
Thanks!

closed 2019-05-16 by InsanityAutomation