InsanityAutomation/Marlin #27: Creality Dwin text corrections.

BoatrightTBC, 2019-03-31:
The thermal runaway bmp has a grammer error, it says that the temperature varied "to much" which should be "too much" 

The english autohome screen says "please wait minutes." which is just wrong.  I would suggest either "please wait a few minutes" or even better "please wait until complete."  

Would you accept a pull request if I did those edits?

InsanityAutomation, 2019-04-07:
Sure, weve fixed a few of the graphics from creality but more are always welcome!

BoatrightTBC, 2019-04-08:
Is there a repository for the DWIN files?  I can’t find one.

Rick Boatright

TBC Software
3410 SW Van Buren STE 202
Topeka, KS 66611

Phone: (785) 272-5993

BoatrightTBC, 2019-04-08:
https://vocshop.com/rick/creality_screen_fixes.zip

includes: Thermal Runaway, Autohome, and a blank bmp with just the background assembled from the visible bits of various other screens.  😊

Rick Boatright

TBC Software
3410 SW Van Buren STE 202
Topeka, KS 66611

Phone: (785) 272-5993

InsanityAutomation, 2019-06-23:
Ive triple checked I have these in B7 so im closing this out!

closed 2019-06-23 by InsanityAutomation