InsanityAutomation/Marlin #29: Unable to Compile for CR-X, MKS Gen L & TMC2208's

OptimalGrammar, 2019-04-13:
Set X, Y, Z, E0 & E1 to TMC2208, press upload, get errors about updating configuration.
#error "[AXIS]_IS_TMC2208 is now [AXIS]_DRIVER_TYPE TMC2208. Please update your Configuration.h."

**Expected behavior:** 
Firmware compiles & uploads to board.

**Actual behavior:**
Error messages.

[Archive.zip](https://github.com/InsanityAutomation/Marlin/files/3076814/Archive.zip)

Zip file attached.

InsanityAutomation, 2019-04-17:
None of the branches here are configured for 2208 in UART mode. Are you trying to maintain use of the touchscreen with the new board and tmc drivers?

OptimalGrammar, 2019-04-18:
That is what I was hoping for, as I'm currently using an Ender 3 LCD with the CR-X as a workaround but it's not ideal. If I configure the 2208's as Standalone, will that work?

InsanityAutomation, 2019-08-24:
FYI, with the port to EXTui, running thos on other boards has become alot easier. Itll get even easier once this gets vetted out enough for me to be comfortable submitting this upstream.

closed 2019-08-24 by InsanityAutomation