InsanityAutomation/Marlin #32: CR10S PRO - Bltouch V3

mennovanhout, 2019-05-12:
### Description

Just installed the latest firmware sadly enough my bltouch v3 still doesn't stop when he hits something.

### Steps to Reproduce

1. Install BLTouch
2. Update LCD and printer firmware
3. Zhome and hold your hand underneath the bltouch

**Expected behavior:** That the printer stops lowering the z axis

**Actual behavior:** Keeps lowering.

chunter1, 2019-05-13:
I had the same problem.
I measured the ouput signal of my BLTouch and saw, that it never gets high.
The solution for me was to connect a 1kOhm resistor between the BLTouch output and 5V to pull the outout (it is an open drain output) high.
Maybe this issue can also be solved by enabling an uC internal pullup?
My board is a Creality3D v2.4 Motherboard with an Atmega2560. (CR-10s Pro).

leseaw, 2019-05-13:
https://www.antclabs.com/wiring3-1

chunter1, 2019-05-13:
@leseaw Is this capacitor also labeled C7 on the CR-10s Pro motherboard?

mennovanhout, 2019-05-13:
> @leseaw
> Is this capacitor also labeled C7 on the CR-10s Pro motherboard?
> Or is there even the schematics of the board available?

No it is not. Also this page is about v1 mainboards howerver 99% of cr10s pro's have the V2.1 mainboard. there is 1 capacitor near the z- input and that is c48. But the question is will this fix it? I don't dare to try it.

chunter1, 2019-05-13:
On my Creality3d v2.4 motherboard, the capacitor is labeled C48 and it's a 100 nF type.
Additionally there are a 10k pullup resistor and a 10k series resistor (i guess) going to the GPIO.

mennovanhout, 2019-05-14:
> On my Creality3d v2.4 motherboard, the capacitor is labeled C48 and it's a 100 nF type.
> Additionally there are a 10k pullup resistor and a 10k series resistor (i guess) going to the GPIO.

I removed c48 yesterday. Didn't solve it. I'm going to add 1k resistors today.

chunter1, 2019-05-14:
I will fit the 100nF capacitor again since it does not have a noticable effect on the measurements but increases noise rejection/false triggering.
Since i am using the additional 1k pullup (10k || 1k), i get a time constant of the RC circuit of 91 us.
To trigger the poitive logic level, it takes roughly 2 x Tau = 273 us which is perfectly fine.
In the opposite direction, the "relevant" one for measuring, it takes even less time since the internal resistance of the BLTouch output is much lower and discharges the capacitor way faster.

mennovanhout, 2019-05-14:
> I will fit the 100nF capacitor again since it does not have a noticable effect on the measurements but increases noise rejection/false triggering and ESD protection.
> Since i am using the additional 1k pullup (10k || 1k), i get a time constant of the RC circuit of 91 us.
> To trigger the poitive logic level, it takes roughly 2 x Tau = 273 us which is perfectly fine.
> In the opposite direction, the "relevant" one for measuring, it takes even less time since the internal resistance of the BLTouch output is much lower and discharges the capacitor way faster.
> Later today, i will hook up my scope to verify the signal edges. :).

I can't solder good enough to reinstall the capicator :(.
But I'll add a 1k resistor today between the red and white wire and post my findings here.

mennovanhout, 2019-05-14:
1k Resistor is not working for me. The BLtouch gets way to less power now. The blue light is almost out and the pin won't come out..... what now?

chunter1, 2019-05-14:
I just measured the BLTouch output signal and it ranges from 0 to 2V only.
Don't know yet why it does not get pulled up to 5V...

mennovanhout, 2019-05-14:
Okay so I got mine to work perfectly.
I removed the c48 capacitor and added a 1k transistor between the red and white wire. This boosts the white wire with 5v from the red wire which solves the problem @chunter1 .

I have to resolder this: But it was a quick test and works perfectly:

[image]

chunter1, 2019-05-14:
Fine to hear, but is your high-level on the signal line also limited to 2V only?

chunter1, 2019-05-14:
And this is what my rising and falling edges look like:

[image]
[image]

chunter1, 2019-05-14:
Very interesting thread:
https://github.com/MarlinFirmware/Marlin/issues/13345

chunter1, 2019-05-14:
Not connecting the BLtouch output to the Creality V3.0 motherboard gives the correct 5V logic signal with clean edges. :/

mennovanhout, 2019-05-14:
> Not connecting the BLtouch output to the Creality V3.0 motherboard gives the correct 5V logic signal with clean edges. :/

Probably the capicator that is eating your power. I've got a new problem i can't put the z offset lower then -0.52 ?? it jumps back ...

chunter1, 2019-05-14:
There is definately a configuration issue because the Z-MIN input pin is pulling low.
As far as i have seen the signal is routed to GPIO PD3 of the Atmega2560.

chunter1, 2019-05-14:
@mennovanhout 
ATTENTION!
Do not use a low resistor value!
I tried 330 Ohms to get reliable triggering 
[image]
and killed my BLTouch!
Luckily only the two schottky diodes were defect and i could replace them :)
So there is definately a problem with the configuration setup.
The GPIO seems to be pulling low and thus we hardly can get the minimum voltage level to sense the line as high!
Maybe some developer can help us with the right configure.h settings?

mennovanhout, 2019-05-15:
> @mennovanhout
> ATTENTION!
> Do not use a low resistor value!
> I tried 330 Ohms to get reliable triggering and killed my BLTouch!
> Luckily only the two schottky diodes were defect and i could replace them [...]

My bltouch is working fine but have some issues with the z-offset it can't get low enough.
Nice to hear that you could fix your BLTouch! I'm a developer myself but not familiar with Marlin. I did a few tests yesterday changing some settings in configuration.h but did not yet succeed to fix the offset issue.

FanDjango, 2019-05-15:
IF:
There is a series resistor (often this is: 1k), a pull-up resistor (often this is 10k), and a capactitor externally on the GPIO pin
AND:
You measure 5V on the signal free floating, but 2V only when connected to the controller
THEN:
Perhaps, due the configuration the GPOI pin itself is configured into "PULL DOWN" mode. It SHOULD be configured into PULL UP mode.
CHECK:
configuration.h. this section:
```
// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_ZMIN_PROBE
#endif
```
At least that is a place to start...

FanDjango, 2019-05-15:
Concerning the capacitor:
Many users report that removing has helped them. Especially when the capacitor was 5uF or more.
100nF is not really an issue, in my opinion. ANTClabs still states: remove it.
The main issue is the current sourcing capability of the BLTouch output from the 5V / 3.3V rail to produce a logic high on the controller pin AT ALL, regardless of how long it takes to get there - unless the capacitor is so large that the voltage doesn't even reach a high enough level before the pulse is over. See the scope waveforms further up.

FanDjango, 2019-05-15:
I would also implore you to mention which of @InsanityAutomation's firmwares or which other firmware you are using and in which version and to post your configuration files, at least.

mennovanhout, 2019-05-15:
> I would also implore you to mention which of @InsanityAutomation's firmwares or which other firmware you are using and in which version and to post your configuration files, at least.

Do you know how to lower the z-offset in marlin configuration? Since the screen doesn't allow me to turn it lower.

FanDjango, 2019-05-15:
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.3   // Z offset: -below +above  [the nozzle]

I always set that one, need maybe one or two iterations to get it real close. Never need to adjust Z-offset again until you change the printer somehow.

mennovanhout, 2019-05-15:
> #define Z_PROBE_OFFSET_FROM_EXTRUDER -1.3 // Z offset: -below +above [the nozzle]
> 
> I always set that one, need maybe one or two iterations to get it real close. Never need to adjust Z-offset again until you change the printer somehow.

If i do -1 the nozzle gets higher instead of lower somehow? and +1 (1) it gets lower but not low enough. Sometimes it looks like the adjust settings during print won't use that value. But i'll run some more tests today.

chunter1, 2019-05-15:
@FanDjango
Thank you for your support!
I'm on the "Creality_DWINTest" branche.
Here is my configuration file.
As far as i see, the pulldowns are not enabled.

[Configuration.zip](https://github.com/InsanityAutomation/Marlin/files/3182119/Configuration.zip)

chunter1, 2019-05-15:
I checked the signal path again and i again come to the conclusion, that there is a 10k pullup to 5V and a 100 nF to GND directly connected to the Z_MIN input pin.
Between the CPU und the input there is another 10k.
So i am totally confused where that low impedance could come from.
I assume there are no compnents placed on the bottom side?

[image]

mennovanhout, 2019-05-15:
@chunter1 Remove c48, I've spoken with antclabs they say no mather what version you have it is always a good idea to remove c48 on cr10s pro.

chunter1, 2019-05-15:
Is the "-Z" labeled signal on the ribbon cable connector the same as the Z_MIN signal?
If so, there is a chance that the problem is located on the interface board where the feeder is mounted.
I can't check now since i am not at home...

[image]

chunter1, 2019-05-15:
Looking at the ribbon cable breakeout board, i guess i have found the cause of the problem.
As it looks there is an optocoupler and a LED connected to the "-Z" / Z_MIN signal line?
Have to measure later to be sure... :)

[image]
[image]

InsanityAutomation, 2019-05-15:
There is another z min input on the main board. Bltouch needs to be wired there. I have not found a need to remove a capacitor on the pro. a lot going on in this thread that I didn't notice until this morning when some of the email notifications flooded through. I'll read through it shortly.

chunter1, 2019-05-15:
@InsanityAutomation
First, thanks for the effort you put into this project! ;)
Can you tell me where this other z_min input is located?

FanDjango, 2019-05-15:
Looks like you are on the right track

mennovanhout, 2019-05-15:
> There is another z min input on the main board. Bltouch needs to be wired there. I have not found a need to remove a capacitor on the pro. a lot going on in this thread that I didn't notice until this morning when some of the email notifications flooded through. I'll read through it shortly.

Do you know how I can lower the nozzle while printing? For leveling everything works perfect but while printing the nozzle is to high since I use the v3 bltouch. And the LCD won't let me put it lower. minimum z offset reached?

chunter1, 2019-05-15:
I disconnected the ribbon cable from the beakout board and well... the signal level is at 5V as it should be.
The question is now, where the alternative Z_MIN input is?

InsanityAutomation, 2019-05-15:
There is only one on the main board itself, just the one on the daughter board cannot be used. The board diagram above shows this, and it's right next to where the capacitor that was in question is located.

Still running around in circles at the day job, the responses during business hours over here are few and far between...

FanDjango, 2019-05-15:
[image]

chunter1, 2019-05-15:
@FanDjango
You mean the socket or the right pin?
Because the right pin is directly connected to the 5 V supply.

FanDjango, 2019-05-15:
If I may add something: The breakout board at the end of the ribbon cable was a great convenience when you wanted to connect a 12V inductive sensor to the printer controller - on less well endowed printers you needed an optocoupler so as not to destroy your controller. Here they have added this for your convenience but it has no sense in using it for a BLTouch.

I mean of course that is the socket you need

FanDjango, 2019-05-15:
One of those pins is 5V, one is GND, and one is trigger-signal in ZMIN.

chunter1, 2019-05-15:
Correct, from left to right:   Z_MIN / GND / +5 V
But i discarded to use this pin now since it is connected in parallel to the long ribbon cable and the voltage limiting breakout board.
I'm gonna try the a real clean solution using the Z_MAX pin instead.

FanDjango, 2019-05-15:
I actually think you can use ZMIN and at that point of the circuit, it will not be interfered with by the "breakout board" and ribbon cable. I prefer ZMIN, but you can also go for ZMAX, no problem. Need to configure this though

FanDjango, 2019-05-15:
This is what the situation looks like, more or less:
[image]

chunter1, 2019-05-15:
> This is what the situation looks like, more or less:
> [image]

Additionally there is a BLUE led connected to +5V in series with a resistor.
I expected the circuit to have absolutely no influence on the signal aswell but somehow it definately gets limited when the ribbon cable is connected to the breakout board.

Now i am struggeling with configuring the Z_MAX input. :)

chunter1, 2019-05-15:
I now connected the BLTouch to the Z_MAX input but it does not work correctly.
When i send G28 and then M48 to do the repeatability test, the BLTouch works as expected and the probing passes.
However if i select on the touchscreen to home, the BLTouch does not releas it's pin to probe but keeps it retracted.
Looking at the servo signal - it is present, but it does not change.
I guess i missed something in the configuration?
Here are my changes since switching from Z_MIN to Z_MAX input:

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN 19

#define USE_ZMIN_PLUG
#define USE_ZMAX_PLUG

FanDjango, 2019-05-15:
Found something on the internet where all said it worked:
If you are using the latest marlin (at least 1.1.4, I think) then yes.. you can do that. You will need to connect the probe to one of the unused pins - Z-max by default in 1.1.4, but make sure to read the comments in the config file - and then make a couple of changes in the configuration file. Look under [Endstop Settings]. Make sure [#define USE_XMAX_PLUG] is not commented out. Depending on your probe, either enable or disable [ #define ENDSTOPPULLUP_ZMIN_PROBE]. Then, under [Z Probe Options], comment out [#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN]. Uncomment [#define Z_MIN_PROBE_ENDSTOP] and I like to uncomment and set [#define Z_MIN_PROBE_PIN X_MAX_PIN] just so I KNOW which pin it's using. Either connect the probe to the X-max pins ore change [X_MAX_PIN] to reflect the set you are using - e.g. [Z_MAX_PIN] to use the Z-max pins. This previous step is EXTREMELY important, so make sure you set that pin correctly. Finally, under [Mesh Leveling], I want to think that you need to comment out [#define Z_SAFE_HOMING]. I don't think that that feature works with the Z probe AND the z-min used together. All of the other probe settings are the same as if the probe IS the limit switch.

chunter1, 2019-05-15:
Guys, thank you so much for your support!
I found the problem and it is on the breakout board.
When nothing is connected to the "Z-" connector, the optocoupler is turned ON leading to a low impedance pull down of the Z_MIN line.
That was the low impedance source i was looking for all the time. :)
**SOLUTION:**
**You must place a jumper on "Z-" connector (as shown below) to keep the optocoupler from pulling down the Z_MIN line.**
Simply using a 1 kOhm or less pullup resistor is NO VALID SOLUTION since it will probably get your diodes inside the BLTouch bunred out one day as mine did immediately (see pics above).

[image]

InsanityAutomation, 2019-05-16:
Ill need to try this on a pre-april 5th bltouch V3, i didnt pay too close attention to the daughterboard at all once I had a detected input signal! Very good observation :)

mennovanhout, 2019-05-16:
@chunter1 good work man

So if you place a jumper there, I can remove my resistor between the white and red wire?

kuksinsky, 2019-05-16:
> Guys, thank you so much for your support!
> I found the problem and it is on the breakout board.
> When nothing is connected to the "Z-" connector, the optocoupler is turned ON leading to a low impedance pull down of the Z_MIN line.
> That was the low impedance source i was looking for all the time. :)
> **SOLUTION:**
> **You must place a jumper on "Z-" connector (as shown below) to keep the optocoupler from pulling down the Z_MIN line.**
> Simply using a 1 kOhm or less pullup resistor is NO VALID SOLUTION since it will probably get your diodes inside the BLTouch bunred out one day as mine did immediately (see pics above).
> With this jumper in place, you don'T need any additional pullup at all.
> 
> [image]

Do I need to desolder a capacitor?

chunter1, 2019-05-16:
> @chunter1 good work man
> 
> So if you place a jumper there, I can remove my resistor between the white and red wire?

Exactly
Did just one run of the probe accuracy test M48 yesterday, but the values are now perfect compared to the pullup mess before.
When i remember correctly, i have approx. 0,002 mm standard deviation now which i think is fine.

kuksinsky, 2019-05-16:
> > @chunter1 good work man
> > So if you place a jumper there, I can remove my resistor between the white and red wire?
> 
> Exactly
> Did just one quick run of the probe accuracy test M48 yesterday, but the values are now perfect compared to the pullup mess before.
> When i remember correctly, i have approx. 0,002 mm standard deviation now which i think is fine.
> 
> @kuksinsky
> I would remove it to get the best performace possible.
> However without the cap you have weaker protection and noise suppression.
> 
> Configuring the BLTouch output to be push-pull may maximize the performance since the rising edge time is no longer depending on the 10 k pullup resister alone but gets actively driven high.
> To profit from this you need to use:
> 
> > #define BLTOUCH_FORCE_5V_MODE // For BLTouch V3.0 force 5V only mode
Thanks for the answer! And one more question, is the correct connection for v3 in this [video](https://youtu.be/PLKEhdznVMY)?

chunter1, 2019-05-16:
> Thanks for the answer! And one more question, is the correct connection for v3 in this [video](https://youtu.be/PLKEhdznVMY)?

Yes, however i don't know why he doesn't use the +5V and GND pin of servo output 0 in one line as they are all the same for each servo.
These are the pins i used since i want to have them all alligned and cleanly installed on one 3-pin connector :)

[image]

chunter1, 2019-05-16:
It's a bit off topic, but i think the settings for bed size and offsets should look like this to avoid crashing into the bed holder clamps.

```
#if (ENABLED(MachineCR10Std))
#define X_BED_SIZE 300
#define Y_BED_SIZE 300
#define Z_MAX_POS 400
#endif

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS -5
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
```

mennovanhout, 2019-05-16:
@chunter1 Are you able to print now? I'm still facing the issue that my nozzle is to high while printing.

chunter1, 2019-05-16:
Did a quick testprint yesterday - worked well.
When the BLTouch nozzle is retracted, how much mm is it above your nozzle?
If i remember correctly, mine is approx. 3 mm above.

mennovanhout, 2019-05-16:
> Did a quick testprint yesterday - worked well.
> When the BLTouch nozzle is retracted, how much mm is it above your nozzle?
> If i remember correctly, mine is approx. 3 mm above.

My nozzle seems to be 7mm to high. what really weird is when I go to movement -> z homeing and then lower the z so it is 0 the nozzle is perfect. But while printing in the adjust settings the z offset is -1.52 and is still way too high? And I can't lower it the LCD won't let me.

chunter1, 2019-05-16:
I followed the mechanical installation guide here:
https://docs.wixstatic.com/ugd/f5a1c8_5a3d7fd6382f4ff5a77fa65b7f4028b9.pdf

mennovanhout, 2019-05-16:
> I followed the mechanical installation guide here:
> https://docs.wixstatic.com/ugd/f5a1c8_5a3d7fd6382f4ff5a77fa65b7f4028b9.pdf

Yo uare using 1.1.9B6 right from this creality_dwinset branch can you share your changes? But not the changes about the probe pin only the settings of z offset etc?

chunter1, 2019-05-16:
> > I followed the mechanical installation guide here:
> > https://docs.wixstatic.com/ugd/f5a1c8_5a3d7fd6382f4ff5a77fa65b7f4028b9.pdf
> 
> Yo uare using 1.1.9B6 right from this creality_dwinset branch can you share your changes? But not the changes about the probe pin only the settings of z offset etc?

I am using the "Creality_DWINTest" branche.
However... since i am completely new to 3d-printing and Marlin, it is still not clear to me what branches are suitable for the CR-10S PRO and which ones are recommended.
Even I'm not sure if the InsanityAutomation repo is the right way to go or use plain Marlin??
Lots to learn yet ;)

These are the settings i remember - i can post the others later today:

`#define X_PROBE_OFFSET_FROM_EXTRUDER -27  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -1   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.1 // Z offset: -below +above  [the nozzle]`

[59 more replies]

closed 2019-07-15 by InsanityAutomation