InsanityAutomation/Marlin #36: CR10S PRO feedrate, acceleration and jerk settings

chunter1, 2019-05-20:
How can the DEFAULT_RETRACT_ACCELERATION  be 1000 when the DEFAULT_MAX_ACCELERATION is set to 25?
Also the DEFAULT_ACCELERATION and the DEFAULT_TRAVEL_ACCELERATION are set to 300 while DEFAULT_MAX_ACCELERATION for the Z-axis is limited to 100?
Are the values for DEFAULT_MAX_FEEDRATE and jerk freely chosen or adapted to the CR-10s pro?
I attached the settings of the Creality sourcecode at the end.

`#elif (ENABLED(MachineCR10SPro))
#define DEFAULT_MAX_FEEDRATE          { 500, 500, 10, 25 }
#define DEFAULT_MAX_ACCELERATION      { 750, 750, 100, 25 }
#define DEFAULT_ACCELERATION          300    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   300    // X, Y, Z acceleration for travel (non printing) moves
#define DEFAULT_XJERK                 8.0
#define DEFAULT_YJERK                 8.0
#define DEFAULT_ZJERK                 0.4
#define DEFAULT_EJERK                 5.0`

From the Creality sourcecode:

`#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 } // { 500, 500, 5, 25 }//

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 500, 500, 100, 5000 }
#define DEFAULT_ACCELERATION          500    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000 //  500  //  // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   500    // X, Y, Z acceleration for travel (non printing) moves
#define DEFAULT_XJERK                 10.0  // 20.0 // 
#define DEFAULT_YJERK                 10.0  // 20.0 //
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0`

InsanityAutomation, 2019-05-23:
There are surprisingly few controls around the actual max values of these... They can be overridden up to 99000 from a graphical lcd despite anything set here. M201 ect can overwrite them to unint32 max... I left some old values there and havnt been over that in awhile, but ill review them again before I finish the 2.0 port of all this.

chunter1, 2019-05-23:
Thank you.
One last question... 
I am a bit confused about what is the difference between these branches and what is the one you are working on for 2.0 release:
```
Creality_Marlin2.0.x
Creality_2.0_Devel
CrealityDwin_2.0
```

InsanityAutomation, 2019-05-23:
Creality_Marlin2.0.x - Stable snapshot tinymachines ships with
Creality_2.0_Devel - synced to head recently standard build for testing (usually stable-ish)
CrealityDwin_2.0 - Active development of touchscreen integration to 2.0

chunter1, 2019-05-23:
Thank you very much for clarification! :)

InsanityAutomation, 2019-06-23:
These have been tweaked a bit more for B7

closed 2019-06-23 by InsanityAutomation