InsanityAutomation/Marlin #37: CR-10S Pro, Micro Swiss Hot End, Thermal Runaway (temperature drops)

quarky42, 2019-05-27:
RE: Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

I was directed to use the Creality_DWINTest branch because of my BLTouch V3 (March build date).  So, I'm not sure that I should or could use the bugfix branch without a lot of modifications.

**My Printer:**
CR-10S Pro using Creality_DWINTest Branch customized and loaded to the printer with the Arduino IDE.
HotEndStock enabled, ABL_BLTouch enabled (V3 / March), AllMetalHotEnd enabled (MicroSwiss HotEnd Installed, stock cooling fans currently but will upgrade those to quieter ones and rerun the PID configuration later.)

I tried disabling "HotEndStock" but it would not compile.  It seems like having AllMetalHotEnd enabled isn't enough.  There are some dependencies on having HotEndStock or one of the other HotEnds Enabled.

I'm pretty sure it is all installed correctly.  I can try tightening the set screws on the heater cartridge portion of the thermal block and make sure they are for sure snug.

The last results of my M303 Auto PID tune were:
Kp 26.66, 2.56, 69.33

### Description

I keep getting the Thermal Runaway Error screen.  Watching the printer, it starts to print PLA at 208 degrees for the first layer.  I have the slicer set to drop the temperature to 205 degrees after that.  The print starts at 208 degrees and after the first layer the temperature drops to 188 degrees before climbing but then it drops again.  

I have re-ran my Auto PID, saved the results, verified they are saved numerous times.  I've tried running it 10x and 5x as both output somewhat different values.

If I manually heat up the hot end using the touch screen, it will get to and sit at what ever temperature I set it to.

### Steps to Reproduce

1. Load the firmware and custom firmware settings described above. Run the PID Auto Tune and save the results to the EEPROM.
2. Slice a simple model where the first layer is at 208 degrees and the other layers are at 205.
3. Print and watch it start printing at 208 degrees, the set point drops to 205 and the Micro Swiss hot end temperature tanks to 188 before starting to recover.

**Expected behavior:** 
I expect it to go to 205 degrees instead of continuing to drop to 188.  

**Actual behavior:**
As stated numerous times:  The temperature drops too much and the system halts during the print with a thermal runaway condition.

#### Additional Information

* Included a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files:
[Marlin-Creality_DWINTest_ConfigurationFiles.zip](https://github.com/InsanityAutomation/Marlin/files/3221768/Marlin-Creality_DWINTest_ConfigurationFiles.zip)

I don't think the Configuration_adv.h was changed from original.  It has a new modified date, but I do believe that was an accidental save with no actual change being made.

I'll go rerun the slicer and just have it print at 205 degrees the whole time and see if that gets me limping along, but I would really like to be able to have a hotter first layer and a little lower temperature for other layers without the printer halting.

InsanityAutomation, 2019-05-28:
Sounds like the part cooling fan is overpowering the hotend heater. Have you tried with the part cooling fan off until quite a bit higher? Or running a PID tune with the cooling fan at 100%?

quarky42, 2019-05-28:
Yes, I have tuned with the part cooling fan on at 50%, 75%, 100%.... makes no difference.   On the latest version of my gcode, the part cooling fan is on at 35% at the start of the print and held at 35% the whole time.   The skirt and the first couple layers are printed at 205 degrees.  The temperature only starts to fall 30 seconds after the extrusion begins.

I can put turn the cooling fan on, preheat the printer to 205, and extrude 300mm of plastic with no loss of temperature.  It is only losing temperature during an actual print job.  I cannot "make it" lose temperature outside of a print job.

quarky42, 2019-05-28:
Also, please note:  If the thermal runaway isn't triggered, then the temperature recovers.  It goes back up to 205 degrees and sits there for 30 seconds.   Then the temperature dives again.   The cooling fan hasn't changed in that time.    As far as I can tell the heater is turning off for 30 seconds at a time. If I am printing slow enough or get "lucky", it will recover.   The print is still ruined with that huge of a temperature swing, but if it does recover, it will go back up to 205 degrees and once it reaches 205 degrees it will sit there for about 30 seconds before it starts to dive again.   Something is causing the heater to cycle in this strange way while printing.

quarky42, 2019-05-28:
Short of having a voltage probe hooked up to the heater line during the print, I am 95% certain that the heater is turning off for 30 seconds at a time.  It is not cycling the way I would expect it to cycle.

InsanityAutomation, 2019-05-28:
Very strange indeed... Are you able to get a serial log when this is occurring? Might be some useful data in the temp reports. One thing to possibly try is disabling autotemp on line 143 in config_adv. It may be having an issue offsetting heater power based on extrusion.

quarky42, 2019-05-28:
I will do both tonight and report back.  If you think of anything else you'd like me to try / check, please let me know and I'll add it to the to-do list.  :)

I can capture the serial output.   I had been using putty to connect.  I finally figured out how to resize the Pronterface window.  The first few times I tried to find the "seem" to resize it, the cursor wouldn't light up.    It wasn't usable until I could stretch it out much wider, finally.

InsanityAutomation, 2019-05-28:
Putty is awesome! Used it for years :)

quarky42, 2019-05-29:
I loaded the B6_250K hex file with factory default PID settings.  I have the same issues, but now the temperature drops are showing up on preheat as well (though I am doing something a bit different now...I got OctoPrint set back up so I've been using that to control the printer. I will need to    So I have graphs, logs, see what the behavior is using the touch screen.)

I haven't had a chance to try loading the firmware from Ardiuno IDE after turning off autotemp in configuration_adv.h yet, but I will try and get to that tomorrow.  I wanted to post the log, screenshots, and a timelapse.

[image]
[image]
[serial.log](https://github.com/InsanityAutomation/Marlin/files/3230964/serial.log)
[headpo_1_20190528232154.zip](https://github.com/InsanityAutomation/Marlin/files/3230966/headpo_1_20190528232154.zip)

quarky42, 2019-05-29:
Should the heated bed be receiving 12 volts or 24 volts?   I took a voltage reading earlier on the DC relay for the heated bed and was reading ~12.3v out of the relay.  Is that right?   

As far as this next graph goes, I think I found my culprit:  The heated bed.  I turned off the heated bed and just the hot end.  The hot end is sitting stable at 205 degrees.   

So my theory at this point is that every time the heated bed is coming on my hot end dips.   *If* (and this is possibly a dumb assumption on my part) the heated bed was supposed to be getting 24V and was instead getting 12V, then it would be pulling a lot more amperage to make up the difference because the bed is heating up to 60 degrees like it is commanded to.   So maybe when the heated bed is cycling on the hot end is cycling off to compensate for the bed draw?

[image]

I didn't see a pot on the heated bed relay, but I didn't look real hard for one.  I didn't put "2 and 2" together until I turned the heated bed off and saw the graph smooth out.

InsanityAutomation, 2019-05-29:
Where were youre leads when you measured 12v? Only one power supply in the system and its 24v... Power supply may have an issue and is failing under load, but only the hotend shows it as the bed is much slower to respond and change.

quarky42, 2019-05-29:
The load side.  It only shows voltage on those connectors when the heated bed is on.  Doing some more reading I think the heated bed is 12V based?

The other side of the relay showed 24.3v constant. (I did not measure that side under load... Sounds like I need to measure the hot end temperature when the heated bed is on, but based on the behavior I am pretty sure the hot end is losing power while the bed is heating.  I am uploading a video that shows the nozzle reach temperature with the bed.  When the nozzle starts to drop the bed drops 1 degree and the nozzle only recovers after the bed has recovered.

InsanityAutomation, 2019-05-29:
Id try to get a replacement power supply. Keep in mind load side of the SSR is +/+ not +/- think of it like a light switch on that side.

juliandroid, 2019-05-29:
It would be interesting if the PSU is throttling (overheating). Inside you have one small fan and if it does not work (if you reverse the polarity it won't run at all)?

quarky42, 2019-05-29:
@juliandroid  Thank you for taking the time to comment. I don't think I got a chance to add this, but yes I already checked the fan last night and it is indeed running.  The fan was pointing into the power supply instead of pointing out of the power supply.  I have it pointing out as an exhaust fan and it has made no difference in the problem at all.

I felt the cover of the power supply after the bed and hot end had been on for a while and it was warm, but not uncomfortable at all.  (So somewhere between 100 degrees and 110 degrees without getting out my IR thermometer.    I had already made sure that all power connections on the power supply and on the control board / dc relay were tight.  They were.  I didn't find any loose connections when I was initially going through the printer to do the quality control that Creality is known to not always do well.

If there is something I should check inside the power supply, I am willing to carefully do so.  At this point I think @InsanityAutomation is spot on... it is likely a weak / bad power supply.

quarky42, 2019-05-29:
What's weird is that I can heat up both the heated bed and the nozzle at the same time.  It takes about ~7 minute to heat up if I heat up both at the same time.  The nozzle heats up a lot faster with the heated bed pre-heated or not running at all.    I do not see the same major drops when the heated bed and the nozzle are going at the same time.

@InsanityAutomation Is the bed heater "all or nothing" or is it varied by PWM in any way?    I haven't run any PID calibration on the heated bed itself.   Is there any value in that?  I'm just using the defaults.   I did load the 250k hex version of the firmware last night and had no change in the behavior.  Since others had confirmed that they used the hex version without problems I figured I would try a "known good" firmware that way to make sure it wasn't something weird in the Arduino IDE project.   I did redownload the github version and I made the change that you suggested a few posts back. I did not have time to load it up and try it.  I am still willing to if you think I should.  I just ran out of time last night.

Here are two videos showing as clearly as I can the temperature fluctuations with the heated bed on and the lack of temperature fluctuations right after turning the heated bed off.  There is only about a second or two between the end of the first video and when the second video starts. I stopped and started recording so that I would have two separate files:
https://www.youtube.com/watch?v=DnFcbnXpij0   (Preheating takes about 7 minutes in this one.  The first temperature dip is a little bit after 8 minutes in.  The second temperature dip is at about 8:24 into the video.  The third is another 2 minutes and change after that.   So a period of a little over 2 minutes between dips)

https://www.youtube.com/watch?v=EZt28lBNxpI
I stopped recording the first video and started this one and turned off the bed.  You can see that for 6 minutes straight the nozzle temperature is stable.   I saw a 1 degree variation on the touch screen at one point.

quarky42, 2019-05-29:
I didn't see a 220 / 110 V switch on the power supply.  Is there one that I missed somewhere?   If so, can someone direct me to it so I can make sure that the power supply isn't on 220V by mistake?

juliandroid, 2019-05-29:
If I remember correctly, somewhere on the Creality site was written that the PSU is automatically selecting the right voltage.

Did you try setting the hotend to something like 240C to see whether time interval between heating and first drop is consistent with lower (205C) or it shorten and more often? I assume that if you have issue with the PSU it will take less time and happen more often if you draw consistently more power on that rail.

Regarding the PSU - I would inspect visually for inflated caps (turned off and not touching anything - for long time there might be 110/220V charges): 
[image]

juliandroid, 2019-05-29:
[image]

quarky42, 2019-05-29:
How obvious would the signs of heating be on the power transistors?   carbon?  severe discoloration?   I didn't notice anything, but I'll look again.

I'll check the caps.  (Yes, I'll make sure not to touch the metallic bits where the angry pixies can leap out at ya!  Still a good warning, though.  ;)    I didn't notice any bulging but I can look for it specifically to be sure.

I have PLA loaded, so I am not going above about 220 degrees on the hot end.  220 just long enough doing the PID tuning causes some of the filament to crystallize.  I suspect the same effect you are talking about would happen if I drove the heated bed temperature up to 80 degrees and/or had something blowing on the bed to cool it down.   It's tied to the bed heater cycling.   My next plan was to watch the hot end voltages when the bed is heating to see what I'm getting when the power supply is under load.

juliandroid, 2019-05-29:
I see at least 2 resettable thermal fuses, but I have no idea why if those kick-in your entire printer is not shutting down. From the datasheet ( https://www.meanwell-web.com/content/files/pdfs/productPdfs/MW/RSP-500/RSP-500-spec.pdf ) it seems it has single rail (output), so everything connected to it should shutdown as well.
Both over power and over temperature protections have automatic recovery, but there is no way for the printer to keep working for 30 seconds. 

The alternative is that only one of the power mosfets is overheating (internal temperature above 150/175C), it shutdowns and leave to the PSU with less power than expected, so the hotend is not completely turned off, but just don't have enough power to keep the temperature steady.

You said you have loaded PLA filament. What about reducing the printing speed significantly, so it becomes quite easy to maintain the hotend temperatures even when the current is reduced due to failing mosfet? I guess you'll see less temp drop than before until the mosfet recovers.

quarky42, 2019-05-29:
Yes, printing at slower speed reduces but does not eliminate the temperature drop. 

With that much information about what might be causing this, I can watch the power supply in FLIR and specifically look okay the mosfets while it's printing and see them over heat in real time.  Your analysis of this situation is ginormously appreciated!

Also, I have a killawatt meter.  I can watch the power going into the printer with the heated bed, the hot end, and then both the heated bed and hot end.  I also looked up the PSU data sheet and with that efficiency rating I can tell if the power supply is drawing more or less power than it should be pulling.  (Which would help rule out or indicate a short or a component pulling way more power than it should.)

We've pretty strongly narrowed this down to hardware and the fact that you're still willing to share your expertise and effort on this is really appreciated.  I may just end up buying a replacement power supply while I wait for Creality to dick around.  Does Tiny Machines sell this particular model?  (I'll check later when I'm back to a computer.)

juliandroid, 2019-05-30:
Btw, I don't recall what was the default fan direction, but from my picture it seems to be an exhaust fan, so you had running your PSU fan in wrong direction and might caused failure. I'm using Noctua fan and I'm not even sure it has enough static pressure and flow compared to the stock, so that could be a potential problem for me (and others performed the mod too), but at least I had it running in the right direction.

I Amazon I saw a lot of "MEAN WELL RSP-500-24", so if you don't find that at Tiny Machines.

Please keep us posted with your findings here.

InsanityAutomation, 2019-05-30:
Forget the box fan, run it submerged in liquid nitrogen! Maybe resting on some dry ice! Meanwell is usually a decent brand so it shouldnt be an issue sticking with them.

quarky42, 2019-05-30:
@insanityautomation this guy gets it!  Go big or go home!  Lol.

I used my wattage meter.  The system pulls between 10 and 14 watts idle.  During bed heating only it pulls between 320 and 400 watts.  (Since resistance goes up with heat and it takes more power to heat up to higher temperatures this makes sense to me.).  During nozzle heating only it pulls about 50 watts (which makes sense for a 40 watt nominal heater cartridge).

Here is where it got interesting: With both the bed and nozzle heating it still only could pull 400 watts.  After running for a little while, it could only pull 360 watts max (thermal limiting).   So when the bed was heating, it took the whole 360 watts.  When the bed stopped heating the nozzle could catch up and it was pulling 50 watts during that time period.

So yeah, pretty sure a bad power supply.  

@juliandroid:
Having the fan blow in would not cause a failure in a good power supply.  Air flow is air flow and this power supply has thermal protection built in.  So it would protect itself.  It might not run optimally with a wrong fan direction, but this one doesn't run any differently with the fan pointing the other way.  I also read the same data sheet that @InsanityAutomation read.  It shows the fan direction on there.  It is definitely in the exhaust direction.

One concern I have: The heated bed is taking not quite twice the wattage it is rated for.  The bed is a 12v 18A if i remember correctly from yesterday.  With just the bed heating up only, the system is pulling up to 400 watts.  If I replace the power supply, how much will the bed pull, then?  Will it burn out?  Is it pulling too much?

Maybe the bed has a failure that has shortened the power supply's life.

juliandroid, 2019-05-30:
Yes, the SSR switches 24V, but after initial load it just maintain the set temperature.
One thing that is strange - you said, you have power meter and the power rating for input is 100-240V AC 5.9A. So, for my power country that is 230V*5.9A ~ 1360W and the DC output is only  24V*21A (~500W). Not sure where that difference is coming, but shouldn't be from efficiency. Anyway, the main point here is that on the outlet side I would expect to see significantly more than those 400W on the AC side, reported by  you when you heat both.
I'll try myself and see what wattage is reported :)

quarky42, 2019-05-30:
At first all I came across was 12v heated beds for the CR-10S.  Now I am starting to see some 24V ones.

This one shows 24v and 220 watts:
https://m.banggood.com/Creality-3D-24V-220W-235235mm-Aluminum-Heated-Bed-Hot-Bed-Kit-For-Ender-3-3D-Printer-p-1335217.html

So I was mistaken about the voltage. (I wondered why they would have a 24v PSU and a 12v bed, but that's all I found at first).  But I still think the wattage is way too high on the heated bed:
https://m.banggood.com/Creality-3D-24V-220W-235235mm-Aluminum-Heated-Bed-Hot-Bed-Kit-For-Ender-3-3D-Printer-p-1335217.html

Is there any way to limit the heated bed to 220 watts from the controller side?   I doubt it.  I figure the bed just gets 24V applied to it and that's it.  However if that's the case then why does the heated bed pull more power the hotter it gets?  (If resistance is going up as it heats up and it is getting the same 24V, then the current and wattage should be going down?  I'm setting guessing my thoughts on that from before.)

juliandroid, 2019-05-30:
I've just checked behaviour of my PSU.
I've setup the hotend to 205C and bed to 60C.
The power consumption (on the AC side) maxout at 505W. It then drop slightly to around 500W and when the temperature became closer and closer to the target it gradually dropped to around 450W (heating according to the PID).
So, most of the time it was in the range of 470-500W.

quarky42, 2019-05-30:
Regarding expecting to see more power being drawn: exactly!

New CR-10S Pro's come with a big aluminum heat sink on the DC Relay now.  So, mine already has that to help it shed heat.

Back to the wattage:. Exactly!  I would expect it to be able to pull up to 500 watts for a good power supply.  This one is supposed to be 89 percent efficient.  So that's a little under 450 watts usable power.  Figure about 15 watts overhead for the controller, fans, etc (based on what I saw). That is about 435 watts for all things (motors, bed heater, nozzle heater).  

Since I'm getting up to 395 watts maximum, draw, that is 365 watts usable (after conversion loss).  I've seen the bed take a lot more power than I think it is rated for.  The bed is starving the nozzle heater.  The videos show it pretty clearly, too.

Definitely not getting the full power out of the power supply.  When the bed heater is on by itself the printer pulls between 360 and 395 watts.  When both the bed heater and the nozzle heater are on, it still only pulls 360 watts (likely because the power supply has heated up and has even lower capacity at that point).  So seeing the hot end temperature drop while the bed heater is active is making more sense.

(Fully agree with the earlier estimates that the power supply is faulty... But I wanted to try and back it up with measurements based on the tools I have available.)

I'm thinking if a Mean Well 600W power supply will fit in there, I'll put that in.  They are 20mm taller.  I'd rather not run a power supply at full load.

juliandroid, 2019-05-30:
Sounds reasonable :) Can you please upload an image of the SSR and the aluminum heat sink? I'm interested whether I can further improve my cooling of the SSR due to the planned switch to 220V AC heatbed :)

quarky42, 2019-05-30:
Sure, I'll post a picture back when I get a chance.  In the meantime I managed to get the video clips uploaded and put into a playlist showing the power draw of my printer at different points:  
1> Initial Bed Heatup
2> A little while later, the bed is at 36C and climbing,  
3> Bed is at 54 degrees and climbing to 60C (So it is near the end of the heatup cycle)
4> Bed has reached 60C, is at steady state, so then I start heating up the hot end to 205C and we watch that heat up all the way as well as see what happens when the bed cycles.

I really like this little wattage meter.  Probably one of the best "a cheap tool to have on hand in case I need it" purchases I've made.

Here's the playlist:
https://www.youtube.com/playlist?list=PLBpIws7wZqQGts1Dw50CgzbvVv3JygCIn

quarky42, 2019-05-30:
@juliandroid Here is that picture of the SSR with the aluminum heatsink.

I really wish Creality had rotated the heatsinks 90 degrees... both that silver one and the black ones.  If they were rotated, then the fans would cool the heatsinks down even better.

[image]

quarky42, 2019-05-30:
@InsanityAutomation It sounds like the power draw isn't correct.  @juliandroid saw 500 to 505 watts being drawn on their system.

I do stand corrected on the voltage of the heated bed, however the 24V heated beds I find online all say 220 Watts.  That heated bed is drawing 315 or more watts by itself.   In @juliandroid's example of 505 watts that would be:

505 Watts * 0.89 efficiency = 449.5 Usable watts
449.5 - 15 watts (LCD, LEDs, Fans, Control Board) = 434.5 watts
434.45 - 40 watts heater cartridge = 394.5 watts.

So based on some number crunching I would say that the heated bed needs somewhere between 385 and 395 watts by itself.   (The heater cartridge is rated for 40 watts and I saw mine use about 40 watts pretty consistently, but it could have been 45 watts or maybe even 50.  The bed, by far, uses a lot more power.)

According to the specs of the power supply it should be able to run up to 130% of its capacity, if I'm reading it correctly.    Still, running at 500 watts on a 500 watt power supply does seem a little crazy to me.

quarky42, 2019-05-30:
Looks like there is around  54mm height available in that base or less.   So being able to run a 600 or 650 watt power supply in there is not possible.  Damn.   Another 10mm or 20mm would be ideal.   If Creality made the base 15mm taller, then we'd have the option of switching power supplies easily.

Hmm...make a 10mm aluminum frame shim all around the base for a taller power supply?  Hmm...I could do that with extrusions.   

I guess if I did the AC bed upgrade using a DC/AC Solid State Relay, that would free up a ton of overhead. I could probably run everything else on 100 watts pretty easily.  :/   Hmm.

juliandroid, 2019-05-30:
@quarky42 Thanks for the picture! I also see grey fan mount blowing they relay - is that printed by you or also stock now?

Yep, the PSU is rated for 504W, 24V DC max, so with 89% efficiency that makes everything except for the motors that already consumes 89% of the maximum. With all of the motors an extra 25W and you are looking at 95% load :)
From the datasheet, the AC input is 2.65A, 230V ~ 609W which related to the rated 504W make rather smaller number, so I guess there is something like extra 5% reserve, but in general it is OK for a reputable PSU to be used on 100% - after all it should work in the rated temperatures/environment and load.
Personally I prefer having something like 50% backup... and I'll achieve that by replacing the heating bed with 230V version :)

chunter1, 2019-05-30:
> @juliandroid Here is that picture of the SSR with the aluminum heatsink.
> 
> I really wish Creality had rotated the heatsinks 90 degrees... both that silver one and the black ones. If they were rotated, then the fans would cool the heatsinks down even better.
> 
> [image]

Thanks for the funny image of the heatsink prooving that nobody at Creality has the slightest clue of thermal management. :)
My printer came with the SSR mounted in the air, not touching anything.
I am using a thin layer of thermal conductive foil between the SSR and the printer case. ;)

quarky42, 2019-05-30:
A big chunk of aluminum is better than stand offs which is what they were using.  

One YouTuber noticed the relay in air and mounted it to the aluminum extrusion rail using a couple T nuts for it.  Even that was enough to cool the SSR down a great deal.

So in this case I think thermal mass is enough, but since they could get heat sinks cut and which size and direction, they should have had the fins turned 90 degrees so that the air flow would be better.

The double fan bracket seen there is my design inspired by a single fan bracket I saw online.

juliandroid, 2019-05-30:
@quarky42  You can contact Keenovo, they are offering 310x320mm silicone pad for CR-10s Pro (110/220V versions). I don't see currently that offer in Ali, but they offered the same size previously for the price of the "normal" CR-10S :) And yes you need DC/AC SSR. The one in the printer is not appropriate :)

quarky42, 2019-05-30:
@juliandroid yeah I agree about running power supplies at around 50 percent load.  They run a lot cooler yet still within their efficiency range at around there.  For me it is more like 50 to 60 percent ideal...70 percent is okay.  Anything over 80 percent is just asking for the power supply to die.  I haven't found a power supply yet that I felt was good for 100 percent load 100 percent of the time.  :)

Looking at the heated bed upgrade.  That's about $70 plus relay and insulation mat.   That's not too bad.  I want to get this power supply replaced and maybe then I'll do that next once I make sure what I bought initially works. Once it's fixed, fix it some more until it's broke... Or something like that.

quarky42, 2019-05-30:
@juliandroid correct.  The SSR in an older version used to be installed on posts, no fan no heatsink.  

Recently they added a heatsink turned the wrong direction still no fan.  This is the version I got.   I added the second fan.

I see the heated beds on Amazon.  Not willing to go that far until the power supply situation is settled, but will keep it in mind.

juliandroid, 2019-05-30:
BTW, adding just a small fan in front of the heat sink will help a lot, since now the fins are pretty much blocked to dissipate heat :D 

Yes, that is discouraging - all the stupid mistakes made by Creality are countless now :)
One comforting news is that peak power consumption is rather temporary, not constantly. Also this can be mitigated by Creality by heating the bed first, then the hotend. Also they can disable LCD for saving extra 1W :wink: and even put the MCU in sleep mode :rofl:

chunter1, 2019-05-30:
Maybe of interest...
Starting at the yellow line, the diagram below shows the power-consumption of my CR-10S PRO printing a calibration cube at 40 mm/s.
Extruder is at 200°C, bed at 50°C and turned off at layer 10.

[image]

juliandroid, 2019-05-30:
That is due to the Eco mode in the printer's menu?

chunter1, 2019-05-30:
> That is due to the Eco mode in the printer's menu?

No, i am using Simplify3D and control everything from there.
By the way... If you don't have it already... i can highly recommend using a Raspberry Pi with Octoprint.
It's really increadible convenient :)
Gonna integrate the Pi into the Creality case next.

juliandroid, 2019-05-30:
Yep, I'm having an older Pi that I want to use it for OctoPrint precisely :) So many things I want to do, so less time to do them :)

quarky42, 2019-05-30:
The inside of my power supply and the DC relay when the heated bed has reached 60C:

[image]

quarky42, 2019-05-30:
That inductor coil though... 70.8 degrees C.  Yeah, it's getting a little too hot in there at 360 to 395 watts of power draw.

juliandroid, 2019-05-31:
From temperature side, I don't see nothing imminent that could cause the problem. Maybe it is time for PSU replacement. I cannot imagine that the problem is coming from the heatbed (both to consume less than rated and also stop working periodically), although for example I've seen sensors working perfectly while cold (normal resistance) that goes to "infinite" when heated and that cycles over and over again :)

quarky42, 2019-05-31:
Yeah I've reached out to the seller, Creality, and Mean Well with the information.  The PSU has a 3 year warranty and Mean Well has a USA based website at least so maybe they can do a warranty exchange in a reasonable amount of time.

I did notice some clicking (rapid cycling on/off) of the solid state relay though.   I think it had more to do with the PSU reaching it's faulty overload limit.  The SSR relay led and the power led on the PSU both blink rapidly for a few seconds.  I'll see about getting that video loaded.

At this point I've just been adding info to the thread because I'm able to do a bit more testing and in the hopes that once I get a replacement PSU that the information here ends up helping someone else that comes across it.

I appreciate the thoughtful discussion that everyone has contributed to this far!

InsanityAutomation, 2019-05-31:
Rapid blinking is expected in pwm control as it gets close to temp. PID takes over within 10c of setpoint.

quarky42, 2019-05-31:
Solid State Relay and PSU LEDs rapidly blinking when the bed gets up to temperature:
https://youtu.be/cQzmc1puYko

From the video it looks like just before the relay turns off, it starts to cycle rapidly.  Does this hint at a control board problem?

I don't believe seen this relay is turning on... Just when it is turning off, so far.

I'll see if this behavior goes away after the psu is replaced.

quarky42, 2019-05-31:
@insanityautomation it is happening only in the last second or two of bed heating.  Not the last 10C.

Also it is causing the power supply LEDs to cycle in time with the SSR.  So is this another indication of a weak PSU?   

Do your PSU LEDs blink just before or as the set point is reached?

quarky42, 2019-06-09:
Creality has been most unhelpful and replies only to messages they feel like it.

I provided what I feel is adequate proof that there is a problem with the power supply.  But that wasn't good enough for them.  They want to see the voltages drop.  So while I'm getting setup to measure the hot end voltage I notice that the power supply dc wires are mirror images of each other.

Would someone be willing to share a picture of their working cr-10s pro power supply connections?

I'll edit this post with the picture of mine when I get to my computer.  

What I'm seeing is that the 24v exhaust fan is on pin 1 and 6.  The bed is on pin 2 and 5.  I assume the hot end is on 3 and 4.    Either it doesn't matter what pins the go on as long as the positive and negative polarities are respected or maybe there is an issue with having the ground return to the wrong pin relative to the positive.  (Not having wired any power supplies before I would assume that pin 1 goes with pin 4, 2 with 5 and 3 with 6, but maybe that isn't necessary.)

juliandroid, 2019-06-13:
Looks like the same to me. Do you have any improvment? New PSU?

[image]

chunter1, 2019-06-13:
> Either it doesn't matter what pins the go on as long as the positive and negative polarities are respected or maybe there is an issue with having the ground return to the wrong pin relative to the [...]

The order of the connections does not matter at all.
Just check if the supply provides constant 24V in every printing stage.

quarky42, 2019-06-13:
Thank you VERY much for the sanity check.   Glad to see it is the same.   Finally the seller AND creality agreed to send a new power supply.  It only took roughly a month to convince them from when I was first convinced (with the help of this thread and everyone contributing to it) that there was a power supply problem.

My "sunshine parting the clouds of confusion moment" was likely when someone confirmed that their unit pulled 500 - 505 Watts with the same settings while mine pulled a max of 390 watts with a cold start and later 360 watts when it had been running for a while....   That right there was a dead give away that the power supply couldn't handle the load (when added to the fact that the hot end by itself pulling 40 watts plus the ~10 watts of overhead worked fine without the heated bed running).   

Later on I even measured the PSU voltage while the heated bed was on:   9 - 13V while the heated bed was active....  yeah that PSU is failing under load.   Still even after all this information they still wanted more proof.   Ugh.   Now to wait the 3 weeks for them to ship me a new one.

I went ahead and ordered a 750W silicone 310mm AC heater and a brand name SCHNEIDER ELECTRIC DC controlled AC relay (40A max), and bed insulation.   I'll may have that conversion done this weekend (and if not then next weekend for sure) and the new power supply will just be a backup for when this one finally eats itself.   Maybe my weak PSU will last long enough at 50 watts output to actually get some prints while I wait for the slow boat from China to arrive.    

My only other challenge is that I would like to somehow AC current limit the bed to 6 amps maximum.   If I do figure out a good way to do that, then I should be able to drive that bed up to a maximum of 200 degrees (really I'm looking for between 130 and 150 degrees, but if I can make it safe for the heater's max temperature of 200 degrees, that provides the kind of margin of safety I like to design for.)   I'm going to use double insulation on the bottom of the bed AND a layer of insulation on top of the printer housing below the bed to keep the heat directed upward.   

I'm intending to eventually design an enclosure that sits over the top of the printer as well.  The design of which would leave the base outside / below the enclosure so that it would still draw in cool air and keep the internals cool.   If/when I do this, I'll remote the extruder motor out using the Zesty Nimble to get the extruder motor out of the chamber.  I might have to put fans on the Z and X axis motor to keep them from heating up too much and/or drive them slower when printing at higher temperatures.    I intend to print in PETG mostly, so that won't be a problem, but I would like the ability to print in POM and/or experiment with high temperature materials.  I already have the all metal hot end for it.

I am really thankful for everyone's contributions to this thread.  I can't imagine how long this would have taken to get Creality to send a new PSU without this help!

juliandroid, 2019-06-13:
If the goal is safety you can go another way:
For example there are temperature limiting switches/fuses: https://www.aliexpress.com/item/KSD9700-250V-5A-15-155-Degree-Bimetal-Disc-Temperature-Switch-Normal-Close-Closed-Thermostat-Thermal-Protector/32860802842.html?spm=a2g0s.9042311.0.0.5b7b4c4doCsbeN

This is for 230V@5A max, which theoretically is way above those 750W. If you pick 155C Normally Closed version then, going above 155C will cut the power. Obviously you have to place it between between the bed and the pad.

quarky42, 2019-06-13:
@chunter1  yep, our replies crossed paths   (more like I was still writing mine while yours came in).   I'm caught up with that understanding.   It had to be either / or.  ;)   I'm just glad to have a new power supply coming my direction finally, and in the mean time maybe an AC bed will work better anyway.  (Even if I was going to the AC bed, I would want a good / proper power supply because I might use it for something else.)

quarky42, 2019-06-13:
> If the goal is safety you can go another way:
> For example there are temperature limiting switches/fuses: [...]

I follow the logic there.  The one problem with that is if the limiting switch / thermal fuse loses contact with the bed, but I could see using both a current limiting device and a thermal switch.   Nothing wrong with safety that would require multiple failures before catastrophic failure occurs, especially considering these kinds of components aren't that expensive.

quarky42, 2019-06-13:
Ok, maybe I'm a geek, nerd or whatever, but I thought this was really helpful / neat to read in relation to my current limiting concerns:  
http://www.resistorguide.com/ptc-thermistor/

So maybe a 7A equilibrium PTC thermister (or maybe a little under 7A like 6.6A because I don't think I'll ever need the full 200 degrees of the pad... maybe around 180 degrees max).

juliandroid, 2019-06-13:
I don't thing those PTC thermistors are intended for high current, high voltage. What you really need is just a normal (ceramic) fuse and fuse holder. Let's say your pad is consuming 6.2A, then I would go with 8A slow fuse and if short hapens this is the first thing to blow. However, as I menitioned before you have to mandatory ground the heat bed directly to the ground lead coming from the power cable. This is more important, due to the nature of the pad - it is just a large resistor/wire that if it breaks it will just have very large resistance, but if that touches the metal plate it might electrocute you if there is a proper grounding.
An extra life saviour could be a WiFi temperature/fire alarm that will turn on and also send you notification.

[16 more replies]

closed 2019-07-04 by quarky42