InsanityAutomation/Marlin #38: Creality_DWINTest Does not compile "'z_values' was not declared in this scope"

eficker, 2019-05-29:
### Description

Configured for my printer's specific upgrades using UBL and several changes. Try to compile but get the following error:

sketch/Creality_DWIN.cpp: In member function 'void RTSSHOW::RTS_Init()':
Creality_DWIN.cpp:289:28: error: 'z_values' was not declared in this scope
       rtscheck.RTS_SndData(z_values[xCount][yCount] *10000, AutolevelVal + (showcount-1)*2);
                            

### Steps to Reproduce

1. Modify configuration.h 
2. Try to compile
[Conf.zip](https://github.com/InsanityAutomation/Marlin/files/3230330/Conf.zip)

This could be an error on my part, but I can't seem to track it down. Any help is appreciated.

InsanityAutomation, 2019-05-29:
UBL is not compatible with the 1.1.9 build. It will be in 2.0 though!

eficker, 2019-05-29:
Awesome! Thanks for the quick response. 

Is DWINTest the best / only branch that currently works with the CR-10s Pro stock touchscreen?

InsanityAutomation, 2019-05-29:
Only stable one... 2.0 is making steady progress but isnt there yet.

eficker, 2019-05-29:
Great, thank you! And thanks for all the hard work on dev. :)

closed 2019-05-29 by eficker