InsanityAutomation/Marlin #39: DWIN to accept broader range of temperatures for the hotend and heatbed

juliandroid, 2019-06-01:
For people with "all metal hotend" might like to have an option from the DWIN LCD interface to change the hotend temperature to 300C or for those with modified heatbed to set temperatures to let say 130 C.

To that end you'll need a modified BMP image (with GIMP, the modified file should be exported as BMP, and from Compatibility Options -> Do not write color space information; and from Advanced Options set -> 24 bits). This file *might* need to be updated in case that files was also modified by Creality:
[61_E_NozzleHotbet.zip](https://github.com/InsanityAutomation/Marlin/files/3243825/61_E_NozzleHotbet.zip)

The more complicated modification however is changing the "13触控配置文件.bin". According to the documentation: https://datasheet.datasheetarchive.com/originals/crawler/dwin.com.cn/f1bae716867596a118366610ee66d00b.pdf 
Section 7. Touch Config. File (13.BIN) and sub section 7.2 Variable Data Input (0x00)
You have to modify address 0x32 (V_min) and 0x36 (V_max) in order to change the maximum (and/or minimum) of those temperatures. The .bin file does not feature any checksums, so you can modify the binary file by directly replacing it via hex editor.

For example, searching for 260C limit:
`FE FF 00 00 00 00 00 00 01 04`
and replacing it with:
`FE FF 00 00 00 00 00 00 01 2C`

and searching for 110C limit:
`FE FF 00 00 00 00 00 00 00 6E`
and replacing it with:
`FE FF 00 00 00 00 00 00 00 82`
does the magic.

Having those two files modified from the DWIN_SET directory/firmware should be enough, once you have updated the screen firmware.

[image]

juliandroid, 2019-06-01:
[image]

[image]

InsanityAutomation, 2019-06-01:
You dug quite deep into that screen! Ill look at making these changes and testing it on this end sometime in the next few days, kids birthday parties and such eating this weekend...

juliandroid, 2019-06-01:
I'm uploading the "original" plain background (no controls or texts on it) that can be used to properly recover the background below the old 0~260C / 0-110C.
[E_Background.zip](https://github.com/InsanityAutomation/Marlin/files/3244260/E_Background.zip)

What I've done for my tests was rather a hack of the original image and also used a standard font to write the new values.
With the blank background image the missing pixels can be recovered properly. Also the China version could be fixed.

juliandroid, 2019-06-01:
Okay, I figured out that it will be easier for me to update both English and Chinese versions with the background mentioned above and a bit better font.
[Updated_NozzleHotbed.zip](https://github.com/InsanityAutomation/Marlin/files/3244280/Updated_NozzleHotbed.zip)

Here is the GIMP src files, so someone could also work on them easier too.
[GIMP_src_NozzleHotbed.zip](https://github.com/InsanityAutomation/Marlin/files/3244282/GIMP_src_NozzleHotbed.zip)

juliandroid, 2019-06-02:
@InsanityAutomation I've created a simple Java app to update the the 13*.bin file:

https://github.com/juliandroid/dwintemps

InsanityAutomation, 2019-06-02:
Wow, you went all out on this! Thanks, looks like good work!

juliandroid, 2019-06-03:
@InsanityAutomation I might have even better news. Please take a look at https://github.com/juliandroid/DWIN_CR_10s_Pro

Here I've put my efforts to recreate and rebuild the DWIN LCD firmware for CR-10s Pro.
The project is easily recreated (most of the deployed artefacts /DWIN_SET/ directory is almost everything needed), except for the "project" file itself and the config file. So extra effort is needed to verify that everything works correctly. However I'm not willing to test it on my own printer (I guess it won't brick, but still - more eyes are needed to look and play with, before going from "experimental" to "beta").

InsanityAutomation, 2019-06-03:
Awesome! sounds like between what both of us are doing, the Marlin 2.0 release supporting this and new touchscreen files should make some very nice upgrades to the machine! Ive got one other project biting at me before I get back to this one, but worst case ill be able to spend some time on it next weekend (no kid events scheduled atm!)

juliandroid, 2019-06-05:
I have updated the project above, also recreated the files again, this time for the latest screen firmware. I'm a bit more optimistic, but still there is one major open question there to be reviewed. Maybe it is ok and is the way it works :)
I'm curious when you have time to test it yourself :)

juliandroid, 2019-06-06:
@InsanityAutomation I've created a new branch "Pro_V1.60.3" at the https://github.com/juliandroid/DWIN_CR_10s_Pro
I've used the original firmware as a source and also replaced both images and changed the Touch layer (via DGUS GUI tool) to reflect the changes of the maximum temperatures for the hotend and heatbed (300C/130C).
I've uploaded that firmware to my printer and seems everything to work correctly!
So my job here is done ;)

InsanityAutomation, 2019-06-23:
Since B7 will include this, im going to close this issue.

Thanks again for the great work over there and documentation on this!

closed 2019-06-23 by InsanityAutomation