InsanityAutomation/Marlin #44: CR-10S PRO ABL working at all?

chunter1, 2019-06-13:
Is the ABL of the CR-10S PRO running latest "Creality_DWINTest" firmware really working right?
Here is what i did to calibrate and what is the result when printing a 0.1mm test pattern:

1. Erased the eeprom with M502 M500
2. Leveled both Z-axis steppers
3. Heated bed to 50°C
4. Did the manual (aux) leveling using the 4 bed screws
5. Turned off ABL on the Screen
6. Started the ABL measurement
7. Turned on ABL on the screen
8. Printed the test-pattern and got shocked :)

Are the ABL correction values from the terminal output (M501) somehow ok with the ones shown on the LCD (see attachments).
What am i doing wrong here?

[image]
[image]

juliandroid, 2019-06-17:
I have some distant memory that might be incorrect:
1. those settings are not stored in eeprom currently;
2. values were populated in wrong order (for example shifted at 90 degrees)

wpte, 2019-06-17:
Does that mean it's required to do ABL before every print instead of retrieving the values?

juliandroid, 2019-06-17:
I don't know, I don't use DWINTest branch. What I saw there is that all writtings to eeprom are commented out, so for example auto bed leveling icon in the LCD menu is lost on power down.

Take a look at this ticket: https://github.com/InsanityAutomation/Marlin/issues/22

wpte, 2019-06-17:
> I don't know, I don't use DWINTest branch. What I saw there is that all writtings to eeprom are commented out, so for example auto bed leveling icon in the LCD menu is lost on power down.
> 
> Take a look at this ticket: #22

Okay, what's the recommended "stable" branch then?

juliandroid, 2019-06-17:
From the product page https://www.tinymachines3d.com/pages/10spro it leads to Creality_DWINTest branch :)

InsanityAutomation, 2019-06-18:
Whats on the screen is skewed after a restart, one of the things 2.0 will fix. Did the test pattern include M420 S1 after G28 in the start script?

chunter1, 2019-06-18:
Yes, it's included in my simplify3d start script:

`...
G28 X0 Y0 Z0 ; home X/Y/Z axes
M420 S1 ; use existing ABL data from eeprom
...`

InsanityAutomation, 2019-06-23:
One of the issues with Bilinear is that its not as simple to adjust individual probe points. 2.0 will have UBL functioning. I did verify that its indeed working as accurately as the probe reads with an intentional tilt of 2mm.

closed 2019-06-23 by InsanityAutomation