InsanityAutomation/Marlin #45: Difference in hex files?

ipatch, 2019-06-17:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

## Difference in two of the **hex** files in the project root of this repo

I noticed there are two hex files that appear to be firmware for the **cr10s pro**

```shell
CR10SPro_Stock_B6_115k.hex
CR10SPro_Stock_B6_250k.hex
```

Both files appear to be ~ 424K, where as they different shasums, and did not see any mention of what the difference is of the two files on this repo?  If any, what is the difference between the above mentioned files?

juliandroid, 2019-06-17:
I think those differences are mentioned on Tinymachines website.
But the difference is in serail interface baudrate, which is important depending of what slicer and configuration you are using.
Variants are 250000 and 115200

InsanityAutomation, 2019-06-18:
Cura seems to have issues with 250000 baud rates sometimes where 115200 they seem stable. Anything else is fine with the faster 250000 baud.

closed 2019-06-18 by InsanityAutomation