InsanityAutomation/Marlin #47: CR-10s Pro - Filament change doesn't work

PrEzi3, 2019-06-30:
### Description
When using the Tiny Machines firmware (the one with BLTouch but I don't tink it matters) plus latest Firmware + Screen Files (B7 / 1.70)  in combination with latest Cura (earlier also don't work correctly) and trying to use the change filament functionality (or the separate plugin -> links will follow) 
it doesn't seem to work - printer just hangs at this step.
I believe that either some functionality is not enabled or screen files are missing for this extra step.

### Steps to Reproduce

1 Download the plugin from here >
https://www.thingiverse.com/thing:2077884/
2 install it into scripts directory e.g. ->
C:\Users\USERNAME\AppData\Roaming\cura\4.1\scripts
(the actual instruction is wrong saying it needs to be put into plugins/subdir/scripts -- at least for cura 4.1.0)
3 Start Cura
4 Load an object into the slicer
5 go to 'extensions' -> 'post processing' -> modify g-code 
6 if you correctly installed the plugin it should appear in the drop down as "change at height 3.4' -> select it
7  set the pause type -> e.,g. layer, change method -> select either M25 M0 or M600 (none works properly), set the pause at layer (e.g. 3)
8 Slice an object 
9 print

**Expected behavior:**
 Printer parks the head at the set position, pauses the print, in the perfect scenario - is able to retract the filament and beep. Then waits for the user to interact (load new filament, press a continue button) then wipes the head, returns to the last print position and resumes print.

**Actual behavior:** 
Either of the commands (M600 M25 M0) should work, but they don't. Printer homes at park print head parameters but doesn't respond afterwards.

#### Additional Information
Even with an unmodified Configuration.h file it still happens.
Tested on the firmware b6 (with old screen files) and also the new b7 + bltouch (and screen files 1.7) -> same behaviour.
I assume that the screen numbers 81 - 83 could be used for that + eventually also 54 and 76/77 to continue...

InsanityAutomation, 2019-07-01:
These need to be resumed by the console with M108 as the screen does not support them. This is being worked on with Marlin 2.0 and was one of the driving elements behind the development of Host Prompt Support.

PrEzi3, 2019-07-01:
> These need to be resumed by the console with M108 as the screen does not support them. This is being worked on with Marlin 2.0 and was one of the driving elements behind the development of Host Prompt Support.

Thank you! 
Does it mean that it will eventually be integrated into the 'CrealityDwin_2.0' branch?
Or which branch of 2.0 are you working on?
Is the firmware already flashable via Arduino or does it still need some work?
Lots of questions... sorry :-)

InsanityAutomation, 2019-07-01:
Yup, thats where itll end up. Still alot of things not implemented there. Its more of a complete rewrite and im going slow to process everything going on instead of just bug hunting. That takes a long time around the day job...

closed 2019-07-02 by PrEzi3