InsanityAutomation/Marlin #48: Missing CR-X hex files

LMS0815, 2019-07-03:
### Description

Missing CR-X files

### Steps to Reproduce

Please provide the 
CR-X_1.1.9_xx.hex and
CRX_BLTouch_ProScreens_1.1.9_xx.hex

files.

PaulF8080, 2019-07-04:
Build scripts normally build things like hex file only when t's dependencies change. The CRX directories didn't change so the hex files should be good.

InsanityAutomation, 2019-07-15:
Ive been tweaking some CRX things over the past week. Ill probably make new hex files for that along with support for the Max soon. All the recent changes have to do with leveling for the most part so not really applicable to a stock CR-X

LMS0815, 2019-07-15:
Thank you @InsanityAutomation. I am using the 3D Touch already. ABL tweaking is exactly what I am looking for.
Current version (CRX_BLTouch_ProScreens_1.1.9_B5.hex)  has issues with OctoPrint receiving errors.

LMS0815, 2019-07-16:
Is there any chance to set the  ABL probe offset via GCode (seems to be firmware hard coded)? 
Unfortunately, I do not know how to create a hex file on my own.

@InsanityAutomation
I like to adjust the probe offset according to this holder...
https://www.thingiverse.com/thing:3438291

Of cause I would be happy if you can share your BT/3D-Touch mount.

LMS0815, 2019-08-16:
> Ive been tweaking some CRX things over the past week. Ill probably make new hex files for that along with support for the Max soon. All the recent changes have to do with leveling for the most part so not really applicable to a stock CR-X

Sorry to bother you... Any progress to share for CR-X?

InsanityAutomation, 2019-08-16:
The 2.0 builds are mostly working, which is a vast improvement for the CRX. Ill be working on it some more this weekend. 2.0 completely automates the tool change process, no more slicer scripts needed, even for a purge bucket.

LMS0815, 2019-08-19:
> The 2.0 builds are mostly working, which is a vast improvement for the CRX. Ill be working on it some more this weekend. 2.0 completely automates the tool change process, no more slicer scripts needed, even for a purge bucket.

Wow. I probably should print a  [_purge bucket_](https://www.thingiverse.com/jnsanders1983/about) now :). I am looking forward to it and hope you can share is soon.

InsanityAutomation, 2019-08-24:
FYI, 2.0 builds are posted, HEX files for CR-X included. Im closing this as resolved :)

closed 2019-08-24 by InsanityAutomation

LMS0815, 2019-08-26:
> FYI, 2.0 builds are posted, HEX files for CR-X included. Im closing this as resolved :)

THANK YOU. Kindly share the link (I probably search the wrong folder)

InsanityAutomation, 2019-08-26:
> 
> 
> > FYI, 2.0 builds are posted, HEX files for CR-X included. Im closing this as resolved :)
> 
> THANK YOU. Kindly share the link (I probably search the wrong folder)

Heres the updated branch - https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

LMS0815, 2019-08-26:
Thank you. Please send me the link.