InsanityAutomation/Marlin #49: CR-10S Pro bed size

darknode, 2019-07-03:
Why the bed size values in the firmware are different from physical size of the bed? It looks like nozzle could move outside of the bed by X-axis because of that.
```
#if (ENABLED(MachineCR10Std))
#define X_BED_SIZE 315
#define Y_BED_SIZE 310
#define Z_MAX_POS 400
#endif
```

Do 5x5 ABL mesh logic has some corrections to calculate correct positions on the bed? (I mean equal offsets by X on the left for 1 point and on the right for 5th point)

InsanityAutomation, 2019-07-03:
Nozzle moves outside of the bed. Probe does not. Firmware knows the difference in distance from the probe to the nozzle.

darknode, 2019-07-03:
I'm just trying to distribute ABL points more evenly across the bed, that's why I'm asking about X Max of 315 and not 310.

[image]

ostehovelen, 2019-07-05:
> Nozzle moves outside of the bed. Probe does not. Firmware knows the difference in distance from the probe to the nozzle.

I'm using the CR-x and this mounting for probe https://www.thingiverse.com/thing:3438291, my probe goes outside the bed, how can i correct this, and where? :O

ostehovelen, 2019-07-05:
I've tried the probe offset, it doesnt seem to do anything....?

zyxep, 2019-07-14:
> Nozzle moves outside of the bed. Probe does not. Firmware knows the difference in distance from the probe to the nozzle.

I'm curious why write the bed as 310x315x400 when it's actually is 300x300x400 ?
I get it with the probe but still

InsanityAutomation, 2019-07-15:
I just put a tape measure across mine, and X measures 310mm, where Y measures 320mm. Nozzle itself when Y is at the limit isnt quite off the bed, so physically the slightly larger dimensions match up. The probe points go as wide as the X gantry can reach. It cant go farther right without crashing. IIRC we left a couple mm buffer to account for manufacturing tolerances. The points are spread out as much as they can be to encompass the largest possible area.

InsanityAutomation, 2019-08-24:
With UBL in there is even more differences people will notice, but for now im closing this. More work will be put into denser meshes in the future.

closed 2019-08-24 by InsanityAutomation