InsanityAutomation/Marlin #51: Stop Button causes bed or nozzle to jerk instead of raising the z position.

quarky42, 2019-07-04:
### Description

I am using the DWIN_TEST branch of my CR-10S Pro (AC Bed, All Metal Hot End, BLTouch V3, timing belt mod) pulled down on June 30th.

When I stop a print (likely because I don't like how the first layer is printing or I realize I forgot to change something important in the slicer) the bed and/or the nozzle gantry jerk about 10mm (estimated).   The nozzle just sits there.  If you are close to the bed, the hot nozzle could leave an imprint on the bed (and mine has because my z offset was off).  Because of the other delays in the firmware after stopping a print and because of how long it takes my fumble fingers to get to the move menu to manually move the z position up 10mm, I would like to request that this be fixed.

I would like to see what ever offset in X/Y that is being commanded be removed and an offset movement in Z be commanded.  If you want to make it REALLY smart, have it check to make sure that we aren't too close to the upper limits.   If someone stops a print with less than 10mm to spare, then just have it move to the max safe height.   In normal situations, just move 10mm up.

### Steps to Reproduce

Stop a print.  Nozzle and/or bed jerk roughly 10mm (give or take, I didn't measure it.  It isn't a big jump, but it is sudden and it is very jerky.)

**Expected behavior:** 

I would expect the nozzle to simply raise up off of the work piece 10mm at most.  (Less than 10mm if the job is stopped near Z-Max)

**Actual behavior:** 
The nozzle jerks to the side in what appears to be a rapid move.  Everything gets jostled.  The nozzle scrapes the work piece.  The nozzle sits resting on the work piece (or if you are at the first layer it sits resting right next to the bed... not good when the nozzle is just sitting there.)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
I don't believe this is applicable.  Please let me know if you still need my Configuration files. I will add them if needed.

* Provide pictures or links to videos that clearly demonstrate the issue.
I don't have any pictures of videos right now.  I need to get back to printing something and I'm having problems with getting the first layer right.  So, I really need to get back to that.  I think this is a very easy to reproduce problem and could be easily seen in the behavior of the stop button.  If I am wrong and you still need a video, let me know, please.

InsanityAutomation, 2019-07-15:
Marlin 2.0 has had stop scripts implemented, so this will likely be covered there. If I keep porting all the features from 2.0 back ill never finish porting the touchscreen forward :)

quarky42, 2019-07-15:
Fair. But to have to nozzle jerk suddenly in both x and y then sit there is pretty bad.  Would really like a more graceful stop.  If I have to wait for 2.0 okay...  Or if there is a simple one or two line fix / workaround that could be safely put in until a better version in 2.0 that would be cool too.

InsanityAutomation, 2019-08-24:
Im going to go ahead and close this with 2.0 out and resolving this!

closed 2019-08-24 by InsanityAutomation