InsanityAutomation/Marlin #52: Pause > Hotend stay on print part.

Fourmi, 2019-07-29:
Hi, i'm on your B7 firmware and when i make a pause ( SD Print), the hotend stay in print part.
Possible to change this, & move hotend to park ( like change filament) or another place ?

Thx

InsanityAutomation, 2019-07-31:
Thats one of the big things weve been working on for the 2.0 port. Its not as simple as it seems unfortunately.

Mikerhinos, 2019-08-18:
Interested in it too, I tried with Cura "change filament" (M600) and "pause at height" (M0), the nozzle moves to where I want, filament is retracted so I can change it, but there's no way to resume print :(

Latest B7 firmware, Cura 4.2.1, CR10S Pro.

InsanityAutomation, 2019-08-18:
2.0 is almost done, and should have something to release shortly on it. M600 works correctly there.

Mikerhinos, 2019-08-18:
Awesome, thanks for the update.
Any ETA ? Like a few days, weeks ?

InsanityAutomation, 2019-08-18:
Were in functional testing now. Initial release within the next week if nothing big jumps out.

Mikerhinos, 2019-08-22:
.hex will be available here ? Yes I'm a bit over excited lol, so many dual color prints ideas...But can't do it because even if I send nozzle to (x = 0 , y = 0) with a "pause at height" Cura post process feature where I want to change filament, and manually pause when it's moving there, it just doesn't stop, gets back on the print, THEN pause :(

InsanityAutomation, 2019-08-23:
Yes, hex files will be uploaded, but source is also here to compile yourself if desired. Do you want to test the current version? Functionality is looking pretty good at this point but more eyes always helps!

Mikerhinos, 2019-08-23:
Sure I'd be glad to test it and help if I can.
Which branch is it ? Because I can't find one newer than several months.
Also is there a tutorial on how to compile ?

InsanityAutomation, 2019-08-23:
https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

Ill grab up some compiling info later today or just put up a preliminary hex. Whats the machine setup? Any bltouch or abl sensor?

Mikerhinos, 2019-08-23:
Stock printer except inductive sensor in place of the crappy stock one. (crappy chinese inductive sensor though haha...). Mirror bed so sensor is installed REALLY close to bed :(

Mikerhinos, 2019-08-24:
Saw that .hex are online, thanks :)
Just downloaded the stock version, there's no need to flash DWIN screen files for this one ?
I'll try to make some tests in the next 5 hours.

InsanityAutomation, 2019-08-24:
> 
> 
> Saw that .hex are online, thanks :)
> Just downloaded the stock version, there's no need to flash DWIN screen files for this one ?
> I'll try to make some tests in the next 5 hours.

There are a few big reasons to! One of them is the addition of M117 messages shown on the top of most screens :)

Mikerhinos, 2019-08-24:
We'll finally be able to display layer number on screen ? Awesome ! Will be really handy when printing in multiple color too !
I can't find the screenfiles archive for 2.0 though :/

InsanityAutomation, 2019-08-24:
> 
> 
> We'll finally be able to display layer number on screen ? Awesome ! Will be really handy when printing in multiple color too !
> I can't find the screenfiles archive for 2.0 though :/

The .7z files are archives (just like zip files, only 70% smaller) with the screen files in em

InsanityAutomation, 2019-08-24:
Keep in mind layer number would come from the host, octoprint ect, Marlin internally has no concept of layer number.

Mikerhinos, 2019-08-24:
I'm printing from USB with Cura because my work computer forces me to encrypt any external media plugged in, so I won't be able to use SD card in the printer if encrypted.

Yes I see the 7zip archives but not one for CR10S Pro, and others are labelled 1.70 instead of 2.0 :/

InsanityAutomation, 2019-08-24:
Thats screen file version numbers (based on the creality 1.70.1 screens)

https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/ProMaxEnd5_1.70_ScreenFiles.7z

This one is CR10S Pro, Max, and Ender5Plus (Anything with leveling)

Mikerhinos, 2019-08-24:
Flashed screen, flashed firmware.
My bed leveling routine is quite weird compared to previous B7, if you guys have a discord channel or something, I'd be happy to share my experience :)

InsanityAutomation, 2019-08-24:
The process is definitely quite different, first off the bed heats first. Then ubl probes vastly differently than bilinear did. Most of the difference is spelled out in the Readme section on the repo main page. Is there things you're seeing different beyond that?

Mikerhinos, 2019-08-24:
Ok so despite this strange leveling behavior, I started a print via USB from Cura 4.2.1.
M117 doesn't show layer number from Cura script, it only show latest message.
M0 still doesn't pause :(
M600 seems to be working sort of though :) It triggers filament change Cura's routine, I had to validate change filament 2 or 3 times, but it worked as intended : totally pause the print, move nozzle to desired position, retract filament, and wait for user input(s) before continuing print on exact desired spot.

InsanityAutomation, 2019-08-24:
I'll peak at the M0 function and see what it's doing. Could look at something for z height shown, but still no concept of layer number in Marlin itself

InsanityAutomation, 2019-08-25:
KEEPALIVE_STATE(PAUSED_FOR_USER);
  wait_for_user = true;

  #if ENABLED(HOST_PROMPT_SUPPORT)
    host_prompt_do(PROMPT_USER_CONTINUE, PSTR("M0/1 Break Called"), PSTR("Continue"));
  #endif
  #if ENABLED(EXTENSIBLE_UI)
    ExtUI::onUserConfirmRequired(PSTR("M0/1 Break Called"));
  #endif

  if (ms > 0) {
    ms += millis();  // wait until this time for a click
    while (PENDING(millis(), ms) && wait_for_user) idle();
  }
  else
    while (wait_for_user) idle();

Looking at the code, M0 should work if called correctly. Ill try to test it sometime tomorrow.

Mikerhinos, 2019-08-25:
Here is what Cura generates : 
`;added code by post processing
;script: PauseAtHeight.py
;current layer: 2
M83
G1 F300 Z1.8399999999999999
G1 F9000 X0 Y0
G1 F300 Z15
M104 S210; standby temperature
M0;Do the actual pause
M109 S210; resume temperature
G1 F300 Z1.8399999999999999
G1 F9000 X155.432 Y177.973
G1 F9000
M82
G92 E354.72931`
It paused for a couple of seconds then got back to printing, maybe CR10S Pro needs a pause time for M0 and default to like 3 seconds.
Anyway for changing filament M600 is way more efficient : it retracts filament, wait for user input on screen, then extrudes 3 massive chunks of filament to purge nozzle before getting back to printing.
I'll try to do more tests with M600 today but it looks like it's working fine and is usable to me :)
There are several "pause at height" in Cura, maybe I didn't used the good one for CR10S Pro and M0 is working (pausing and waiting for user input).

LMS0815, 2019-08-27:
> The process is definitely quite different, first off the bed heats first. Then ubl probes vastly differently than bilinear did. Most of the difference is spelled out in the Readme section on the repo main page. Is there things you're seeing different beyond that?

I can not find anything in the Readme section regarding ABL. 
On my CR-X ABL is evaluating _only_ 16 points, although it claims 25 to be measured.
(I am using the hex you provided)
Is this a bug?

InsanityAutomation, 2019-08-29:
Im going to go ahead and mark this closed, as the initial problem is gone, Can still keep discussion going here or ping me in the discord's in the readme if preferred.

closed 2019-08-29 by InsanityAutomation