InsanityAutomation/Marlin #53: CRX_BLTouch_DualRunout

LMS0815, 2019-08-26:
Thank you for sharing all your work and _sorry for raising an issue_, although I only have a question.
Is there any site, board, etc. where I can find the requirements of the  Dual Runout sensor (switch) and the pins they are connected to?

InsanityAutomation, 2019-08-29:
They are setup to use the X+ and Y+ endstop ports. Disabling the Lerdge option changes polarity if you find it reversed from what you have.

LMS0815, 2019-08-30:
I tried the ports you advised. I probably need to disable the the Lerdge option.
How to do it? I am not able to create a Hex on my own.
Any Gcode to do so?
Any terminal command?

InsanityAutomation, 2019-09-09:
Missed the added question here... Switching the cables is the best route if you are not setup to compile. Ill be posting some guides on compiling from scratch soon though.

closed 2019-09-09 by InsanityAutomation

LMS0815, 2019-09-09:
> Ill be posting some guides on compiling from scratch soon though.

Thank you. Looking forward to it.