InsanityAutomation/Marlin #57: OctoPrint job never ends

LMS0815, 2019-08-30:
After an OctoPrint print job is finished (NOT SD CARD) then Printer display keeps on claiming PRINTING.
I need to abort the job at the printer display to get control back, although OctoPrint is not printing any longer. And status is 100%

The same configuration had no issues with the Marlin 1.x. Therefor I assume this is a bug.

SkullKill, 2019-09-02:
on octoprint, it says 100%, printer physically is all good, (bed and hotend off,).

LCD Still shows 99%, with the stop, pause, adjust buttons. but on top of lcd, it says "TM3D 10S  Pro Ready" "Cool..."

[image]

LMS0815, 2019-09-03:
The octoprint _persistent_ message is a well known cache issue which is will beaddressed in Octoprint [V1.3.12](https://www.youtube.com/watch?v=NwUrvTi9TVc) 
Never the less the CrealityDwin_2.0 branch (CRX_BLTouch_DualRunout_2.0_B1.hex) has an issue not _knowing_, when octoprint finalized a job which is new. V 1.x did not have this issue with Octoprint  V1.3.11. The printer keeps waiting for data and is showing the printing display/status.

InsanityAutomation, 2019-09-07:
Print job timer callback issues fixed upstream and new hex files going up today.

closed 2019-09-07 by InsanityAutomation