InsanityAutomation/Marlin #60: marlin freeze after a print has finished or cancel from octoprint

SkullKill, 2019-09-08:
### Description

using CR10SPro_Stock_UBL_DW2.hex or compiling with custom config

marlin freeze after a print has finished or cancel from octoprint.

when the print finish, the hotend still stays on the last point of the 3d printed object, and just 1/2 crash there. at least the hotend and bed turns off. 
usb comunication/serial does not work anymore at that point. octopi try to reconnect but fails.

### Steps to Reproduce

1. Start a print from Octopi
2. let it finish
3. when it finish, hotend stays on last point of print and freeze, usb/serial communication also stop working.

option 2
1. Start a print from Octopi
2. press cancel
3. usb/serial communication stop working.

option 3 (probably will also happen if i let the print finish, but did not try)
1. Start a print from the LCD , sd card
2. press cancel
3. usb/serial communication stop working.

**Expected behavior:** [What you expect to happen]
usb/serial communication continue to work.

**Actual behavior:** [What actually happens]

usb/serial communication stop working.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

issue was not there 15 days ago, in the previous hex

InsanityAutomation, 2019-09-08:
Found a loop in the print job timer stop, update will be up shortly.

InsanityAutomation, 2019-09-09:
Update is posted, thanks!

closed 2019-09-09 by InsanityAutomation

SkullKill, 2019-09-09:
Thanks, all working now. the screen also clears properly after a print.

FYI UBL with a grid of 8 works fine on the CR-10S Pro. i did try 10, but that crashed it, probably ran out of memory.

p.s 
- only tested printing from octoprint, not tested from SDcard,
- bedleveling done via terminal, with gcodes, did not test from the lcd screen.

Sketch uses 179008 bytes (70%) of program storage space. Maximum is 253952 bytes.
Global variables use 7049 bytes (86%) of dynamic memory, leaving 1143 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.

output of "M100 F" on a fresh boot

Send: M100 F
Recv: 
Recv: bss_end               : 0x1D89
Recv: free_memory_start     : 0x1D89
Recv: free_memory_end       : 0x215B
Recv: Stack Pointer         : 0x215B
Recv: Initializing free memory block.
Recv: 
Recv: 
Recv: 728 bytes of memory initialized.
Recv: 
Recv: Found 728 bytes free at 0x1D90
Recv: M100 F
Recv: fmc() n=978
Recv: free_memory_start=0x1D89  end_free_memory=0x215B
Recv:  (1) found=**728**
Recv:   block_found=1 return=0
Recv: check_for_free_memory_corruption() = 0
Recv: ok P15 B3

output of "M100 F" during/after a print

Send: M100 F
Recv: 
Recv: bss_end               : 0x1D89
Recv: free_memory_start     : 0x1D89
Recv: free_memory_end       : 0x215B
Recv: Stack Pointer         : 0x215B
Recv: Found 328 bytes free at 0x1D91
Recv: M100 F
Recv: fmc() n=978
Recv: free_memory_start=0x1D89  end_free_memory=0x215B
Recv:  (1) found=**328**
Recv:   block_found=1 return=0
Recv: check_for_free_memory_corruption() = 0

InsanityAutomation, 2019-09-09:
Yeah, dynamic memory is lower than I'd like.... I'll get there trimming it down eventually!