InsanityAutomation/Marlin #63: Filament sensor does not work

SkullKill, 2019-09-13:
### Description

branch CrealityDwin_2.0
running on CR-10S Pro

looks like the firmware is not considering the state of the Filament sensor 

### Steps to Reproduce

1. remove filament, sensor light off
2. head up the hotend  > 195 degrees
3. press the FEED filament . it is trying to extrude instead of giving the no filament / change filament message.

OR

1. start a print
2. when the filament runs out, sensor light off, it continues printing

OR

1. start a print without any filament
2. printer still tries to print without any filament

**Expected behavior:** [What you expect to happen]
pause print, and ask to change/load filament

**Actual behavior:** [What actually happens]
continues the print as if the filament sensor has not been triggered.

#### Additional Information

Send: M412
Recv: echo:Filament runout ON
Recv: Filament runout distance (mm): 5.00
Recv: ok P15 B3

LMS0815, 2019-09-14:
I have the same behavior with CR-X (and assumed my cabeling might be wrong.

hdtechk, 2019-09-14:
I had a filament run out last night. I did get the screen pop up.   But extruder had done a retraction some time after sensor trigger and jammed filament back in to sensor when I hit yes to change filament it immediately started trying to print again

SkullKill, 2019-09-15:
> I did get the screen pop up.

@hdtechk which branch are you using? is it the CrealityDwin_2.0 branch ?

https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

hdtechk, 2019-09-15:
Yes I'm using CrealityDwin_2.0

SkullKill, 2019-09-16:
Apologies, this WAS a cabling issue!!!!, i just assume that the cabling was correct, because the led light was behaving as expected. replaced the cable, and it is working as expected!! 
at least output of M119 is....

closed 2019-09-16 by SkullKill

LMS0815, 2019-09-16:
> Apologies, this WAS a cabling issue!!!!, i just assume that the cabling was correct, because the led light was behaving as expected. replaced the cable, and it is working as expected!!
> at least output of M119 is....
> 
> 2 out of 3 wires were working... it just so happen that the sensor wire was the bad one....

Kindly share the right wiring.

SkullKill, 2019-09-16:
@LMS0815 this is on a CR-10S Pro, that already has the filament sensor. it is all already wired from factory, there is a break out board on the gantry to the sensor where it is originally.

for mine, since i converted my setup to a direct drive, i also moved the filament sensor to be on top of the excruder, which is on top of the hotend.... to do that, i had to make a new longer cable, it is a three cable harness, which i just copied from the original cable. basically, pin 1 to pin 1, pin 2 to pin 2, pin 3 to pin 3.

https://www.thingiverse.com/thing:3807893

[image]
[image]
[image]

@hdtechk 
not sure if that is helpful, but the M600 settings (which is used in a filament runout, i think) are in Configuration_adv.h
just under ADVANCED_PAUSE_FEATURE

i changed those settings to suite the direct drive setup on mine.