InsanityAutomation/Marlin #64: Fast Manual Feed Rate when Z-axis in motion

SkullKill, 2019-09-15:
### Description

Very Fast Manual Feed Rate (Pressing the FEED button on the LCD) when Z-axis in motion

### Steps to Reproduce

1. heat up hotend to >195 degrees
2. move z axis up, maybe 30mm
3. while z axis is still moving, press the "FEED" button on the LCD

**Expected behavior:** [What you expect to happen]

Feed at a slow rate

**Actual behavior:** [What actually happens]

it is feeding the filament at an extremely fast rate.

#### Additional Information

when the z axis has finish moving, the FEED button works as expected, feeding slowly.

InsanityAutomation, 2019-09-16:
Sounds like an upstream ExtUI issue, but ill verify and see where it comes from.

InsanityAutomation, 2019-10-27:
Ive confirmed this is resolved upstream between either Marcio's or my changes. Dont know which one for sure, but doesnt matter :)

closed 2019-10-27 by InsanityAutomation