InsanityAutomation/Marlin #65: CR-10s Pro MicroSD Card Refresh

ipodsync, 2019-09-25:
### Description

When removing and re-inserting the MicroSD card, the firmware does not re-read the card.  It shows the same files as before the card was ejected.

### Steps to Reproduce

1. Press print to view files on micro SD card
2. Remove card and add files
3. Re-insert card into printer
4. Press print to view files.  New files are not displayed in list. Reset printer and list updates.

**Expected behavior:** [What you expect to happen]
When print is pressed, list all files on card

**Actual behavior:** [What actually happens]
When print is pressed files present at last re-boot are listed.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AndrzejH, 2019-09-30:
I noticed the same thing.

But this problem doesn't appear when I upload to the printer HEX files from page : https://www.tinymachines3d.com/pages/10spro
- CR10SPro_B7_Stock_115k.hex
- CR10SPro_B7_BLTouch_115k.hex

Then I can change SD card and files are read correctly.
The issue is only with firmware compiled from source code.

AndrzejH, 2019-09-30:
This commit is source of problems for CR10SPRO

file : Update Creality_DWIN.cpp

https://github.com/InsanityAutomation/Marlin/commit/41462c780cdafe7e13c14d88876329a30859a177#diff-3bf220dd50055530cc0f350bb1b5429b

` - const bool sd_status = IS_SD_INSERTED;`
` + const bool sd_status = card.cardOK;`

and another by some reason doesn't work.

https://github.com/InsanityAutomation/Marlin/commit/a7638725e16347ec85c31b391c062179f75796ec#diff-3bf220dd50055530cc0f350bb1b5429b

` - const bool sd_status = card.cardOK;`
` + const bool sd_status = (card.cardOK || IS_SD_INSERTED);`

I reverted that change and after compilation and upload SD Card works fine

I left only:
`const bool sd_status = IS_SD_INSERTED;`

I know that is only workaround, but I don't know that project in details to investigate intention of use "card.cardOK" instead of "IS_SD_INSERTED"

InsanityAutomation, 2019-10-27:
The whole SD file handling is in need of a rewrite top down....

quarky42, 2020-06-03:
I'm having this problem with 2.0 DW6.2 on a CR-10S Pro with a SKR V1.4 Turbo Motherboard.  Having to reboot to see new files is rather painful.

Before I upgraded motherboards the stock board would often see new files, but it too had its own bugs when the list would not be updated.

I do hope that the SD File handling gets sorted.

InsanityAutomation, 2021-04-28:
Im closing this as the SD handling has been completely redone at this point

closed 2021-04-28 by InsanityAutomation