InsanityAutomation/Marlin #66: Printing Cura gcode via OctoPrint missing last commands

LMS0815, 2019-09-26:
Printing cura gcode  with OctoPrint is not properly ending the job.

1. The job is not fully ended (some commands at the end are missing). Therefor the nozzle keeps on sitting on top of the print instead of moving to the "removal position". (_G1 F3000 X0 Y270_ is not executed)

2. OctoPrint is not getting a proper job-end confirmation. After printing is done, the job is marked an not successful (red) instead of successful (green), although no error message is displayed.

```
;FLAVOR:Marlin
;Generated with Cura_SteamEngine 4.1.0
T0
M82 ;absolute extrusion mode
G21 ;metric values
G28 ;home all
G90 ;absolute positioning
M107 ;start with the fan off
G1 F2400 Z15.0 ;raise the nozzle 15mm
M109 S200 ;Set Extruder Temperature and Wait
M190 S60; Wait for bed temperature to reach target temp
T0 ;Switch to Extruder 1
G1 F3000 X5 Y10 Z0.2 ;move to prime start position
G92 E0 ;reset extrusion distance
G1 F600 X160 E15 ;prime nozzle in a line
G1 F5000 X180 ;quick wipe
G92 E0 ;reset extrusion distance
G92 E0

;TYPE:SKIN
G1 F1800 X170.315 Y157.062 E408.93357
G1 X170.335 Y157.062 E408.93407
G1 X170.335 Y164.705 E409.12473
G1 X170.315 Y164.705 E409.12523
G0 F7200 X170.325 Y164.695
;TIME_ELAPSED:947.149574
G1 F4200 E406.12523
M140 S0
M107
M104 S0  ;hotend off
M140 S0  ;bed off
G92 E0
G1 F2000 E-100  ;retract filament 100mm
G92 E0
G1 F3000 X0 Y270  ;move bed for easy part removal
M84  ;disable steppers
M82 ;absolute extrusion mode
M104 S0
;End of Gcode

```

InsanityAutomation, 2019-10-27:
From what I can tell M104/M109 S0 are the culprit. Its in the print job timer stopped event triggers.

LMS0815, 2019-11-07:
The issue was seen in CRX_BLTouch_DualFil_UBL_DW3.hex
Seems to be solved (or not existing) in CRX_BLTouch_DualFil_BIL_DW4.hex

InsanityAutomation, 2020-01-01:
Just wanted to confirm this is completely gone before I close this

closed 2020-04-04 by InsanityAutomation