InsanityAutomation/Marlin #67: Second extruder is turning in wrong direction on the CR-X with dual gear extruder

Maalow, 2019-10-03:
My second extruder (E1) is turning in the wrong direction, because there is no variant of the Creality dual gear extruder for the second extruder (E1) of the CR-X and I had to buy the extruder for the left side (E0) twice. The left extruder (E0) will of course turn in the correct direction. 
Can you add a variant of the hex file that inverts the second extruder motor (E1) and sets both (E0 and E1) to 140steps/mm in the Configuration.h file?

PaulF8080, 2019-10-28:
You have the wrong model.  There is a left and right model that turn in opposite directions.

Maalow, 2019-10-28:
> 
> 
> You have the wrong model. There is a left and right model that turn in opposite directions.

But not from Creality. I bought this one: https://www.amazon.de/dp/B07TZ79SV6/
There are some knock offs but they have a cheaper quality.

PaulF8080, 2019-10-28:
I should have done more detail.  You buy a left and a right extruder. Only one of yours is wrong. The motors are meant to turn in the opposite directions. The settings are proper and not an issue,  in my opinion.

Maalow, 2019-10-28:
> 
> 
> I should have done more detail. You buy a left and a right extruder. Only one of yours is wrong. The motors are meant to turn in the opposite directions. The settings are proper and not an issue, in my opinion.

I think I've been more unclear: **There is no dual gear extruder for the right hand side from Creality.**
I already know that I need one for each specific side, this is nothing new for me. But if there aren't any, what else is left like buying two for the same side and inverting the direction on one side? The printer would do his job again with two color prints. The second extruder just needs the inversion.

InsanityAutomation, 2020-04-04:
Current bleeding code, lines 1741 and 1742 in configuration.h control this invert
#define INVERT_E0_DIR false
#define INVERT_E1_DIR true

closed 2020-04-04 by InsanityAutomation