InsanityAutomation/Marlin #70: Z distances changes between purge and start of print/ ABL readings are'nt being read correctly

cryzzgrantham, 2019-10-29:
### Description-->  ABL is just above nozzle, was work working perfectly on B6, abl measuring 25 points works, but seems to not be reading measurements correctly 

  ABL readings aren't being read correctly 

### Steps to Reproduce

1. [First Step] define ABL_NCSW //Creality ABL or Any NC Sensor
2. [Second Step] set x offset to 44, set y offset to -2

**Expected behavior:** [What you expect to happen] 
purge z height should be the same as when print starts.

ABL Bed level readings should be used to deviate z correctly. 

**Actual behavior:** [What actually happens] 
purge line seems to be ~.2mm too close. 

over the 25 bed level points, half seem correct, a quarter seem too close, a quarter seem too far away

#### Additional Information replaced ABL sensor (non stock inductive), placed sensor just above nozzle to ensure correct/ reliable readings. recompiled B7 to ensure no user errors where made yet abl issues persist.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[configurations.zip](https://github.com/InsanityAutomation/Marlin/files/3785497/configurations.zip)

InsanityAutomation, 2019-11-29:
Found an issue doubling babysteps. Will be resolved in the release update this weekend.

cryzzgrantham, 2019-12-07:
> Found an issue doubling babysteps. Will be resolved in the release update this weekend.

So is my issue fixed now mate?

InsanityAutomation, 2020-04-04:
Im closing this as it appears the issues has long since been resolved

closed 2020-04-04 by InsanityAutomation