InsanityAutomation/Marlin #72: UBL/CR-C: Probe is released to early therefor it "hits" the bed

LMS0815, 2019-11-06:
During bed leveling:
After a mesh point is measured for the first time, the probe is immediately(?!) released prior to moving the head upwards for the second measurement.
Therefor the probe hits the bed.

This behavior is new to DW4.

The probe should be released after the head reached his highest position.

InsanityAutomation, 2019-11-29:
This is a high speed probing mode that Antclabs is supporting for machines with faster z movement.

LMS0815, 2019-12-02:
Seems that Z-movement is not faster on my CR-X. 
Either the High speed method is not supported on CR-X 
or the implementation is suffering from issues.

LMS0815, 2019-12-21:
Same behavior with DW5. Can yo revert to the old - non high speed - processing?

jesusbackflips, 2020-02-12:
-------Configuration_adv.h--------- >>>also try "BLTOUCH_DELAY" 500ms set, fixed mine with 700ms

/**
   * Settings for BLTouch Smart 3.0 and 3.1
   * Summary:
   *   - Voltage modes: 5V and OD (open drain - "logic voltage free") output modes
   *   - High-Speed mode
   *   - Disable LCD voltage options
   */

  /**
   * Danger: Don't activate 5V mode unless attached to a 5V-tolerant controller!
   * V3.0 or 3.1: Set default mode to 5V mode at Marlin startup.
   * If disabled, OD mode is the hard-coded default on 3.0
   * On startup, Marlin will compare its eeprom to this vale. If the selected mode
   * differs, a mode set eeprom write will be completed at initialization.
   * Use the option below to force an eeprom write to a V3.1 probe regardless.
   */
  #define BLTOUCH_SET_5V_MODE

  /**
   * Safety: Activate if connecting a probe with an unknown voltage mode.
   * V3.0: Set a probe into mode selected above at Marlin startup. Required for 5V mode on 3.0
   * V3.1: Force a probe with unknown mode into selected mode at Marlin startup ( = Probe EEPROM write )
   * To preserve the life of the probe, use this once then turn it off and re-flash.
   */
  //#define BLTOUCH_FORCE_MODE_SET

  /**
   * Use "HIGH SPEED" mode for probing.
   * Danger: Disable if your probe sometimes fails. Only suitable for stable well-adjusted systems.
   * This feature was designed for Delta's with very fast Z moves however higher speed cartesians may function
   * If the machine cannot raise the probe fast enough after a trigger, it may enter a fault state.
   */
  #if NONE(MachineCR10Orig, LowMemoryBoard)
    #define BLTOUCH_HS_MODE
  #endif

  // Safety: Enable voltage mode settings in the LCD menu.
  //#define BLTOUCH_LCD_VOLTAGE_MENU

#endif // BLTOUCH

LMS0815, 2020-03-09:
Still existing in DW6. Is there any GCode to switch Mode?

LMS0815, 2020-03-30:
I am not able to compily any version for my CR-X
If anyone can please provide working defines for the Setting files.

InsanityAutomation, 2020-04-04:
Removed high speed mode for the CRX machines in the current bleeding code

closed 2020-04-04 by InsanityAutomation