InsanityAutomation/Marlin #74: Can't compile with UBL enabled

SkullKill, 2019-11-09:
### Description

Can't compile with UBL enabled

### Steps to Reproduce

un comment UBL

`#define ABL_UBL`

**Expected behavior:** [What you expect to happen]

compile without any errors.

**Actual behavior:** [What actually happens]

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp: In static member function 'static void unified_bed_leveling::G29()':

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:341:43: error: no matching function for call to 'onMeshUpdate(const mesh_index_pair&, int)'

             ExtUI::onMeshUpdate(closest, 0);

                                           ^

In file included from sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:49:0:

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:140:12: note: candidate: void ExtUI::onMeshUpdate(uint8_t, uint8_t, float)

       void onMeshUpdate(const uint8_t xpos, const uint8_t ypos, const float zval);

            ^~~~~~~~~~~~

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:140:12: note:   candidate expects 3 arguments, 2 provided

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:141:19: note: candidate: void ExtUI::onMeshUpdate(const xy_uint8_t&, float)

       inline void onMeshUpdate(const xy_uint8_t &pos, const float zval) { onMeshUpdate(pos.x, pos.y, zval); }

                   ^~~~~~~~~~~~

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:141:19: note:   no known conversion for argument 1 from 'const mesh_index_pair' to 'const xy_uint8_t& {aka const XYval<unsigned char>&}'

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:544:62: error: no matching function for call to 'onMeshUpdate(const mesh_index_pair&, float&)'

                     ExtUI::onMeshUpdate(closest, g29_constant);

                                                              ^

In file included from sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:49:0:

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:140:12: note: candidate: void ExtUI::onMeshUpdate(uint8_t, uint8_t, float)

       void onMeshUpdate(const uint8_t xpos, const uint8_t ypos, const float zval);

            ^~~~~~~~~~~~

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:140:12: note:   candidate expects 3 arguments, 2 provided

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:141:19: note: candidate: void ExtUI::onMeshUpdate(const xy_uint8_t&, float)

       inline void onMeshUpdate(const xy_uint8_t &pos, const float zval) { onMeshUpdate(pos.x, pos.y, zval); }

                   ^~~~~~~~~~~~

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:141:19: note:   no known conversion for argument 1 from 'const mesh_index_pair' to 'const xy_uint8_t& {aka const XYval<unsigned char>&}'

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp: In static member function 'static void unified_bed_leveling::probe_entire_mesh(const xy_pos_t&, bool, bool, bool)':

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:786:49: error: no matching function for call to 'onMeshUpdate(mesh_index_pair&, const float&)'

             ExtUI::onMeshUpdate(best, measured_z);

                                                 ^

In file included from sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:49:0:

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:140:12: note: candidate: void ExtUI::onMeshUpdate(uint8_t, uint8_t, float)

       void onMeshUpdate(const uint8_t xpos, const uint8_t ypos, const float zval);

            ^~~~~~~~~~~~

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:140:12: note:   candidate expects 3 arguments, 2 provided

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:141:19: note: candidate: void ExtUI::onMeshUpdate(const xy_uint8_t&, float)

       inline void onMeshUpdate(const xy_uint8_t &pos, const float zval) { onMeshUpdate(pos.x, pos.y, zval); }

                   ^~~~~~~~~~~~

c:\users\skullk~1\appdata\local\temp\arduino_build_381661\sketch\src\lcd\extensible_ui\ui_api.h:141:19: note:   no known conversion for argument 1 from 'mesh_index_pair' to 'const xy_uint8_t& {aka const XYval<unsigned char>&}'

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp: In static member function 'static bool unified_bed_leveling::smart_fill_one(uint8_t, uint8_t, int8_t, int8_t)':

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:1338:48: error: 'pos' was not declared in this scope

             ExtUI::onMeshUpdate(x, y, z_values[pos.x][pos.y]);

                                                ^~~

sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:1338:48: note: suggested alternative: 'pow'

             ExtUI::onMeshUpdate(x, y, z_values[pos.x][pos.y]);

                                                ^~~

                                                pow

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

SkullKill, 2019-11-09:
i assume this is the reason why UBL hex file were not made available?

https://github.com/InsanityAutomation/Marlin/issues/71

SkullKill, 2019-11-11:
UBL compiles fine under the CrealityDwin2.0_Bleeding branch. (but i did not flash the bleeding firmware on the machine.... yet....)
also i can see changes in bleeding to support the screen on skr1.3 .

i guest we just have to wait for the changes from bleeding to be merged to CrealityDwin_2.0 branch

LMS0815, 2019-11-11:
I am still not able to compile any version for my CR-X. Can you provide a link to a how-to?

SkullKill, 2019-11-15:
update to this. i can confirm that UBL works fine under CrealityDwin2.0_Bleeding branch. (Printed fine)

also, modified it to support the SKR PRO V1.1, and submitted a pull request into CrealityDwin2.0_Bleeding.

tested on SKR PRO V1.1 with the I2C EEPROM module (AT24C256), and UBL enabled.

[image]
there were a few issues with this mount, will post this once it is finalized.

[image]
E stepper motor cable was the wrong way in this pic

[image]
still testing, hence why the cable mess. will clean up and crimp some proper cables soon.

InsanityAutomation, 2019-11-29:
Fixed upstream right after DW4 was released. Next build will have UBL again. Should get pushed this weekend.

closed 2019-11-29 by InsanityAutomation