InsanityAutomation/Marlin #79: Hotend set to 215°C after ABL?

chunter1, 2019-11-30:
I got the problem that after finishing an ABL (initiated through the touch display menue) the hotend temperature ist automatically set to 215°C.
Besides that the system seems to freeze until the temperature is reached?

hdtechk, 2019-11-30:
It hasn't finished the ABL yet it heats up hot end so you can set Z offset then when you hit home it's finished

chunter1, 2019-11-30:
If i get you right, it is intended to heat up?
If so, then it shouldn't be that way at all.
All ABL should do is ABL and the user decides manually at what temperature he is doing it.

InsanityAutomation, 2019-12-17:
The intention at this point is a full commissioning. It heats the hotend and prompts to set the z offset as the last step. The hotend must be heated or this step or you will not get an accurate adjustment. If you want to edit this behavior, it runs the script from USER_GCODE_1 in configuration_adv.h

closed 2019-12-17 by InsanityAutomation