InsanityAutomation/Marlin #80: BLTouch 3.1 on CR-10S Pro dont work.

SFreiheit, 2019-12-01:
Printer: Creality CR-10S Pro
Board version is 2.4

We have turned on the BLTouch in the configuration.h.

We downloaded the latest version of Git (Branch: CrealityDwin_2.0). The printer starts but the sensor only blinks!

Do you know where the problem could be?

[Configuration_zip.zip](https://github.com/InsanityAutomation/Marlin/files/3908500/Configuration_zip.zip)

howlingpan, 2019-12-12:
I have the same issue

dnsbrandy1, 2019-12-13:
Hi,you need to change the black and white cables over,that worked for me.

stretch84, 2019-12-15:
Samsies. 
I have cr-10s pro with motherboard v2.4 and bltouch installed, probe self test and doesn't register afterwards. Creality, tinymachines and several other companys tell you to do different things with the two pin wire. One company says to put it on Z+with brown wire towards inside... other company says no put it on Z-
Another says spin it...(?) Brown wire outside? 
Other company will say put in resistor,  another will say take the current competitor out. 
Btw... I have no black and white cable. Just orange and brown.

InsanityAutomation, 2019-12-19:
Either way works, Z+ use the CR10SProV2 toggle, Z- use the original CR10SPro toggle. The BLTouch pin is the only difference between those toggles.

Polarity of the 2 pin wire (black / white from antclabs) is important. White must go to signal, as black is ground and has continuity to the other ground on the 3 pin (so one could be omitted). If using Z- then you need the jumper on the daughterboard.

Im closing this as there is little possibility outside wiring.

closed 2019-12-19 by InsanityAutomation

stretch84, 2019-12-19:
I appreciate the response,  tiny machines got a hold of me first and was
able to put me on the right track and I was able to finally print

patadeloso, 2019-12-22:
Switching the Z wires in the connector fixed this issue on my CR-X w/ BL 3.1 too.