InsanityAutomation/Marlin #85: CR-10s Pro - unable to connect via usb and unable to move Y in the - direction

joro5kof, 2019-12-13:
I am trying to compile 1.1.9 for the CR-10s Pro after installing a DD e3d Hemera. I make amendments in config.h to enable the e3d hotend, all metal hotend, temp to 300, enable bltouch and zprobe offset to match the mount that I'm using. the firmware compiles fine, I upload it to the printer and it is able to boot and all. 

### Steps to Reproduce
After uploading the edited firmware and turning the device on, if you try to home the printer, Y does not move in the - direction. Z and X work both ways.

furthermore, I am able to flash the pre-compiled b7 hex file via USB, however no other application sees the printer -pronterface, cura, octo. If i flash it back to B7 everything works as it should.

my configuration.h and configuration_adv.h
[sss.zip](https://github.com/InsanityAutomation/Marlin/files/3962703/sss.zip)

InsanityAutomation, 2019-12-19:
Most likely the wrong baud rate is selected in the terminal software for however it was configured. The update im planning to finish today with 2.0 for the machine will have a simple toggle for the hemera setup already.

InsanityAutomation, 2020-04-04:
Closing this as with no activity im assuming the issue was resolved

closed 2020-04-04 by InsanityAutomation