InsanityAutomation/Marlin #86: CR-X: CRX_BLTouch_DualFil_UBL_DW5  G29 J creates Firmware ERROR

LMS0815, 2019-12-20:
Sending `C29 J; Grid leveling` crashes the Firmware and Octoprint is loosing connection.

patadeloso, 2019-12-22:
Seems to happen with CRX_BLTouch_NoFil_UBL_DW5.hex too

Send: G28
[...]
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Printer halted. kill() called!"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M104 T1 S0*36
Send: N5 M140 S0*96
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor

LMS0815, 2020-03-10:
Solved in DW6 (CRX_BLT_UBL_Fil_DW6.hex)

LMS0815, 2020-03-20:
solved: DW6

closed 2020-03-20 by LMS0815