InsanityAutomation/Marlin #88: Bed Size of CR-10 Max wrong

Pixelshading, 2019-12-21:
In the Configuration.h the Printbed Size for the CR-10 Max is set to: 

X=475 Y=450 Z=470

but the actual working Printbed Size is:
X=450 Y=450 Z=470.

Because of the wrong Printbed Size the Autoleveling with BL-Touch hits the Right Metal Clamp that keeps the Printbed on the Heated Aluminum Carriage, because of that the leveling measurements are "wrong" .

InsanityAutomation, 2020-01-01:
Ill take a look at it again. I dont have a machine physically here but the positioning was set to allow the head to move the full distance and probe more of the bed than it could if the nozzle was limited to only the bed position.

InsanityAutomation, 2020-04-04:
This was resolved along the list of M851 changes

closed 2020-04-04 by InsanityAutomation