InsanityAutomation/Marlin #89: LCD screen firmware update Fail on CR10S-Pro, now black LCD screen

spzjulien, 2019-12-25:
hello, i did update many time my CR10s pro

but today i got power fail on the house, right after i start LCD screen update
now i have no display on my LCD screen

is there any way for fix that ?
thanks for your help

InsanityAutomation, 2020-01-01:
According to the dwin manuals, there is a serial flash method from a PC however I have not performed it myself. The tools from https://github.com/juliandroid/DWIN_CR_10s_Pro would be necessary to do so. I have not had a failed one in hand, and I have too little hardware to risk bricking and jeopardize the ability to keep making progress unfortunately.

spzjulien, 2020-01-01:
do you know where i can find more detail about uart serial flash for this screen ?

jesusbackflips, 2020-02-10:
Try to rurn off printer and insert the card then not before. also try standard files to flash first and format stick. Make sure its only 8GB big. i had issues with 32GB. 
link stock screen, board files "creality dot com/download/firmware_c0008

closed 2020-04-04 by InsanityAutomation