InsanityAutomation/Marlin #90: Out of filament message after every print

marcinbauer85, 2019-12-29:
### Description
I get an: "No filament click yes to change filament or no to Cancel"

### Steps to Reproduce
1. Print hatever.

#### Additional Information
My configs:
[config.zip](https://github.com/InsanityAutomation/Marlin/files/4008459/config.zip)

Photo:
[image]

InsanityAutomation, 2020-01-01:
Are you able to get a serial log of when this occurs? Can you attach a gcode file of anything it occurs with? I wasnt able to reproduce here.

Guilouz, 2020-01-04:
For me, I have this message with a PID command via Pronterface when it's done.

juanpa669, 2020-02-01:
AS @Guilouz  I got runout filament dialog box after PID with the button Tools->Hotend PID auto tune.
Did you know if your new PID values are stored in EEPROM @Guilouz ?

Hyrules, 2020-03-16:
I have the same message in pronterface with PID and with touchscreen in menu PID autotune.

InsanityAutomation, 2020-03-26:
Applied a fix in the Bleeding branch, can you give it a shot?

Hyrules, 2020-03-26:
I will try this fix. Keep you posted in a day or two.

Hyrules, 2020-03-26:
Ok got time to test and unfortunately this is not working. I still have a No filament error after sending a pid autotune.

SENDING:M303 S200 U1
PID Autotune start
 bias: 77 d: 77 min: 197.37 max: 207.86
 bias: 76 d: 76 min: 196.84 max: 203.57
 bias: 76 d: 76 min: 197.24 max: 203.57 Ku: 30.55 Tu: 28.50
 Classic PID
 Kp: 18.33 Ki: 1.29 Kd: 65.31
 bias: 75 d: 75 min: 197.11 max: 203.57 Ku: 29.54 Tu: 28.68
 Classic PID
 Kp: 17.72 Ki: 1.24 Kd: 63.53
 bias: 75 d: 75 min: 197.20 max: 203.57 Ku: 29.99 Tu: 28.83
 Classic PID
 Kp: 18.00 Ki: 1.25 Kd: 64.85
PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define DEFAULT_Kp 18.00
#define DEFAULT_Ki 1.25
#define DEFAULT_Kd 64.85
ok P7 B3
==onUserConfirmRequired==
//action:prompt_end
//action:prompt_begin M1 Stop
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user

Hyrules, 2020-03-27:
We'll it does not work when sending a command through pronterface but it does not seem to happen when I use the menu on the printer.

InsanityAutomation, 2020-04-04:
If the issue only persists from pronterface id have to see the ful log of whats sent. I cant see anything left machine side that would cause it.

InsanityAutomation, 2020-04-11:
Im closing this as I havnt heard any update. If it comes up again, just reopen the issue.

closed 2020-04-11 by InsanityAutomation