InsanityAutomation/Marlin #95: CrealityDwin_2.0 (DW5) // after pausing print, then resuming -- Extruder stepper off

PrEzi3, 2020-01-20:
Using CrealityDwin_2.0 branch, latest clone which was DW5. Adjusted for Titan Aero + custom placed BL-Touch sensor. (Please note -- my Titan does not have reverted directions, using the Titan-bundled stepper motor, but adjusted the cabling so the revert is not needed).

Steps to reproduce:
Print something.... pause --- head moves up, and homes / bed moves according to the settings.
Change filament appears with yes and no.
Pressing yes does resume the print.... head /bed moves (X Y Z working fine) into the position and starts to "print". Head moves normally etc. 
However Extruder is locked and not moving at all. Even after stopping the print manual feeding/retracting via button in menu doesn't work. 
What only works is restart via cycling power. Then everything is back to normal again.
Funny enough -- injecting M600 via Cura changeatz plugin seems to work fine and doesn't block the extruder.

Config + advances (haven't touched that one) included.
[config.zip](https://github.com/InsanityAutomation/Marlin/files/4088012/config.zip)

PrEzi3, 2020-02-03:
Update --- I am not a developer, but I have checked the 
\Marlin\src\feature\pause.cpp and found the following:

comment line 318 reads that there is a disable E stepper put in place  --> 
found in lines 369 - 373

However I am unable to find the priming of E stepper back after the operation is done.
Might be this issue? Again... not sure... not a programmer myself... just trying to think logically...

tihoangyeudau, 2020-02-04:
Yes. Im using this code. Sometime i stop printinh then my extruder stopped and idle

InsanityAutomation, 2020-04-04:
I havnt seen the issue here but the base code has been updated several times, can anyone confirm if this still occurs?

cuboy29, 2020-04-14:
I am seeing something strange when I abort a  print.  It would abort but then the extruder would continue to extrude but makes lots of noise.  My guess is because the nozzle cooled down so it couldn't extrude.

quarky42, 2020-04-29:
I have tried to pause prints a couple of times now and not had any complete success, just a couple failures and one partial success.

On My CR-10S Pro using DW6
When I pause from the touch screen and try to resume the print, the print may or may not resume, but it eventually fails.  Seems the head stops moving and the file somehow is at the end, but the print isn't complete.

On my Ender 5 Plus using DW6
When I pause in the gcode, the pause occurs, I change the filament to a different color, the screen displays that the filament sensor detects that the filament is out.  I press ok to continue.  The print continues for a little while, but it just ends with the nozzle sitting on top of the print just oozing all over the place.  I have to manually cancel the print and I don't believe it has printed all the layers,   It definitely printed 1 complete layer in the new color and maybe a second layer, but that was it.

So pausing from the control panel and pausing in GCODE itself (pause generated by Cura using the pause at layer function) doesn't appear to be working correctly.

czadikem, 2020-08-28:
I am also having this issue on my CR10sPro both with DW6 and DW6.2

InsanityAutomation, 2021-04-28:
There are several open items related to improving feedback during the M600 process and its one of the more involved items to work on going forward. This needs to be resolved upstream before it can come back here. Im closing most of these to consolidate them into one thread.

closed 2021-04-28 by InsanityAutomation