InsanityAutomation/Marlin #96: Ender 5 Plus homes to wrong location

xeddog, 2020-01-23:
### Description

Using TM3D CrealityDwin2.0_Bleeding branch, Ender 5 Plus using safe homing, homes way off center.

### Steps to Reproduce

Simply home the machine. 

**Expected behavior:** [What you expect to happen]
Should home to the center of the bed at X175 Y175
**Actual behavior:** [What actually happens]
Homes to position X110 Y110
#### Additional Information

In the configuration.h are these lines:

#if ENABLED(E3D_DUALFAN_MOUNT)
  #define HOMING_ADD 15
#else
  #define HOMING_ADD 0
#endif

******************* And these lines ****************

#elif ANY(MachineEnder5, MachineEnder5Plus)
    #define Z_SAFE_HOMING_X_POINT 110 + HOMING_ADD    // X point for Z homing when homing all axis (G28).
    #define Z_SAFE_HOMING_Y_POINT 110 + HOMING_ADD    // Y point for Z homing when homing all axis (G28).

These may be fine for the Ender 5, but not the Ender 5 PLUS.  Home position for the E5+ should end up being  X175 Y175.

jesusbackflips, 2020-02-10:
looks like x y should be  both +/- reverse to move in diagonal direction more to pause, endposition.

jesusbackflips, 2020-02-11:
After back and forward. Needs to be....
#define Z_SAFE_HOMING_X_POINT 180
#define Z_SAFE_HOMING_Y_POINT 180

Because Bedsize is set to 360. With 175 its again way off. Maybe a calcutaion thing?

xeddog, 2020-02-11:
Close counts.  :-)

InsanityAutomation, 2020-04-04:
With the changes to M851 and UBL G29J reasoning for creating a square with home point and 3 points for a least squares plane are no longer relevant. Set to calculate center point now.

closed 2020-04-04 by InsanityAutomation