InsanityAutomation/Marlin #97: CrealityDWIN_2.0 Printing from SD card starts then fails with "Error:SD read error"

klwill1192, 2020-01-30:
### Description
I have added a BL Touch 3.0 and an E3D Hemera hotend/extruder to my Creality CR-10S Pro.  I am able to print successfully through the USB port using Pronterface.  However, I can no longer print from an SD card.  The printer warms up, homes, and prints the purge lines then hangs.  The Pronterface log shows an apparent endless loop of "[ERROR] Error:SD read error" messages.  I have to power cycle the printer to get out of this.

### Steps to Reproduce
I am using the default start GCODE for the CR-10S Pro in Cura.  I added an M111 and a couple M117s, but these are not getting executed.  I open an STL file in Cura, slice it, save the GCODE to an SD card, insert the SD card in the printer, then start the print using the touchscreen menu.

I have tried three different SD cards, reformatted the cards, and tried two different CR-10S Pro motherboards.  I have also tried enabling each of the various options for SPI_SPEED.  I have also tested with both the CrealityDWIN_2.0 and CrealityDWIN_2.0_Bleeding branches.  The problem does not appear to be related to any of these.  

Here is the start GCODE prepended to each model:
;FLAVOR:Marlin
;TIME:11161
;Filament used: 5.87148m
;Layer height: 0.2
;MINX:124
;MINY:129
;MINZ:0.2
;MAXX:176
;MAXY:181
;MAXZ:101.6
;Generated with Cura_SteamEngine 4.4.1
M140 S60
M105
M190 S60
M104 S210
M105
M109 S210
M82 ;absolute extrusion mode
M111 S7 ;Enable debugging messages
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line

M117 Purge complete

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
M117 Start GCODE Complete

Here is the section of the Pronterface log where the error occurs:

2020-01-29 22:32:49,099 - RECV: echo:DEBUG:ECHO,INFO,ERRORS
2020-01-29 22:32:49,101 - RECV: echo:M201 X500.00 Y500.00 Z100.00 E5000.00
2020-01-29 22:32:49,101 - RECV: echo:M203 X500.00 Y500.00 Z10.00 E50.00
2020-01-29 22:32:49,101 - RECV: echo:M204 P500.00 R1000.00 T500.00
2020-01-29 22:32:49,101 - RECV: echo:M205 X8.00 Y8.00 Z0.40 E5.00
2020-01-29 22:32:49,109 - RECV: echo:M220 S100
2020-01-29 22:32:49,109 - RECV: echo:M221 S100
2020-01-29 22:32:49,111 - RECV: echo:G28
2020-01-29 22:32:51,092 - RECV: echo:busy: processing
2020-01-29 22:32:53,779 - RECV: echo:busy: processing
2020-01-29 22:32:54,183 - RECV: ==onConfigurationStoreWritten==
2020-01-29 22:32:55,779 - RECV: echo:busy: processing
2020-01-29 22:32:57,779 - RECV: echo:busy: processing
2020-01-29 22:33:00,000 - RECV: echo:busy: processing
2020-01-29 22:33:02,000 - RECV: echo:busy: processing
2020-01-29 22:33:04,019 - RECV: echo:busy: processing
2020-01-29 22:33:06,010 - RECV: echo:busy: processing
2020-01-29 22:33:08,012 - RECV: echo:busy: processing
2020-01-29 22:33:10,013 - RECV: echo:busy: processing
2020-01-29 22:33:12,012 - RECV: echo:busy: processing
2020-01-29 22:33:12,832 - RECV: X:159.00 Y:155.00 Z:5.00 E:0.00 Count X:12720 Y:12400 Z:2000
2020-01-29 22:33:12,832 - RECV: echo:M420 S1 Z2
2020-01-29 22:33:12,832 - RECV: echo:Bed Leveling ON
2020-01-29 22:33:12,832 - RECV: echo:Fade Height 2.00
2020-01-29 22:33:12,832 - RECV: echo:G92 E0
2020-01-29 22:33:12,832 - RECV: X:159.00 Y:155.00 Z:5.00 E:0.00 Count X:12720 Y:12400 Z:2000
2020-01-29 22:33:12,832 - RECV: echo:G1 Z2.0 F3000
2020-01-29 22:33:12,839 - RECV: echo:G1 X10.1 Y20 Z0.28 F5000.0
2020-01-29 22:33:12,849 - RECV: echo:G1 X10.1 Y200.0 Z0.28 F1500.0 E15
2020-01-29 22:33:12,861 - RECV: open failed, File: /PLR.
2020-01-29 22:33:12,891 - RECV: echo:G1 X10.4 Y200.0 Z0.28 F5000.0
2020-01-29 22:33:12,891 - RECV: echo:G1 X10.4 Y20 Z0.28 F1500.0 E30
2020-01-29 22:33:14,555 - RECV: Error:SD read error
2020-01-29 22:33:14,559 - [ERROR] Error:SD read error

2020-01-29 22:33:14,559 - RECV: Error:SD read error
2020-01-29 22:33:14,559 - [ERROR] Error:SD read error

2020-01-29 22:33:14,559 - RECV: Error:SD read error
2020-01-29 22:33:14,559 - [ERROR] Error:SD read 
… and so on for another 40,000 lines.

I am attaching my Configuration.h and Configuration.adv.h files.  I believe all my changes to the default values are flagged with my initials: [KLW]
 
[Configuration.zip](https://github.com/InsanityAutomation/Marlin/files/4132462/Configuration.zip)

jesusbackflips, 2020-02-13:
i had issues with more then 32GB SD. best was 8GB also to flash. SD format is FAT32? 
Try maybe only 40mm/s print for test. Dirt, broken solder joint?

klwill1192, 2020-02-13:
> i had issues with more then 32GB SD. best was 8GB also to flash. SD format is FAT32?
> Try maybe only 40mm/s print for test or print Resolution to 0,05mm - 0,1 less stress for CPU etc.. Dirt, broken solder joint?

Thanks for taking the time to respond.

I tried a number of SD cards, both 8GB and 32GB, reformatting all of them to FAT32, but it did not solve the problem.  I was never able to get past printing the purge lines in the start GCODE generated by the slicer, so print speed and resolution have not come into play with this problem.

I also tried switching to a 12864 LCD and the TH3D unified firmware, and the SD card read errors went away.

jesusbackflips, 2020-02-14:
Already have seen  reprap DOT org/forum/read.php?146,115453
"The sdrampstest works after I changed "SPI_FULL_SPEED" to "SPI_HALF_SPEED", otherwise it would just say, that it would not detect my FAT partition."

InsanityAutomation, 2020-04-04:
It definitely appears that power loss recvocery was pushing the card over the edge. Ive seen a few cards it just refuses to function on. If my memory is correct Tim still has it off by default so yes by that feature being removed the problem is gone.

closed 2020-04-04 by InsanityAutomation