InsanityAutomation/Marlin #101: BLTouch 3.1 on CR-10s Pro with CrealityDwin_2.0

cchamilt, 2020-02-05:
### Description

When using the BLTouch, the probe fails to drop quite often.  Previous stable Tiny Machines firmware it was quite rare, now I can hardly complete a abl call without the gantry dangerously shooting to the top in fail with each bad probing.

I didn't see a related issue open or closed, nor do I see any new related commits to bleeding.
### Steps to Reproduce

1. Installed BLTouch 3.1 with old stable firmware, probe sticks about 1 out of 1000 attempts (still not great, but OK).
2. I updated to CrealityDwin_2.0.  Both with the standard hotend carrier and a new printed one with custom BLTouch geometry.  Both fail to drop probe very often on abl so that I spend hours trying to finish even one abl.  Happens with hot or cold hotend.

**Expected behavior:** [What you expect to happen]
BLTouch probe drops when/if requested and measurement sequence starts.

**Actual behavior:** [What actually happens]
BLTouch probe fails to drop and goes into error mode.  Usually within 5-10 attempts from power on.

I am using platform.io to build, let me know if you want me to try and patches or config changes.
Thanks!
[MarlinConfigs.zip](https://github.com/InsanityAutomation/Marlin/files/4162497/MarlinConfigs.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

PrEzi3, 2020-02-10:
Hi Mate, I cannot replicate your problem. 
Also using BLTouch (version 3.0) and CR-10s Pro (mainboard v2.4.1 with TMC steppers) without any problems with BLTouch. Zero.
Difference is the version (3.1 vs 3.0), platform.io vs arduino, and also I have applied a jumper on the board that is near the extruder stepper (found it somewhere in one of the recommendations on github I think).
BLTouch is flawless with all the new TinyMachines firmwares based on Marlin 2.0 (on the older ones 1.1.9x rarely happened that the measurement was too fast and bltouch had not enough time to prime itself, but not happening anymore).

cchamilt, 2020-02-11:
I am not sure why it is sticking.  If I catch it in time, I can force the pin to drop when I see it flash and sometimes that resets the alarm too.

I have a buck converter to drive the rpi4 running octoprint and I have swapped fans to 12v noctuas with bucks as well hoping it might fix this.  (The bucks have some diode protection unlike what most people use in printer hacks.)  The creality fans were starting to fail with less 100 hours on them.  I added smoothers to the x and y steppers as well in case they were adding noise into the 5v system.  I also rebuilt the cable to the BLTouch from scratch and it is as good as crimped connectors get.  Nothing really changed.  Other than smoothing all steppers and replacing the power supply and motherboard, I am not sure what else I can blame in the system at this point.

I may try to get the BLTouch replaced as it does stick a little with old marlin, but the behavior seems to be much worse in 2.0.  I am guessing the ABL gcode is a slight bit different in the versions and whatever changes aggravate the BLTouch.  I will try to find diffs for ideas.

jesusbackflips, 2020-02-11:
look for bltouch in files,code under Configuration_adv.h . for me 700ms delay between probes fixed 500ms was std. there are also option for 5v systems

jesusbackflips, 2020-02-12:
-------Configuration_adv.h---------

/**
   * Settings for BLTouch Smart 3.0 and 3.1
   * Summary:
   *   - Voltage modes: 5V and OD (open drain - "logic voltage free") output modes
   *   - High-Speed mode
   *   - Disable LCD voltage options
   */

  /**
   * Danger: Don't activate 5V mode unless attached to a 5V-tolerant controller!
   * V3.0 or 3.1: Set default mode to 5V mode at Marlin startup.
   * If disabled, OD mode is the hard-coded default on 3.0
   * On startup, Marlin will compare its eeprom to this vale. If the selected mode
   * differs, a mode set eeprom write will be completed at initialization.
   * Use the option below to force an eeprom write to a V3.1 probe regardless.
   */
  #define BLTOUCH_SET_5V_MODE

  /**
   * Safety: Activate if connecting a probe with an unknown voltage mode.
   * V3.0: Set a probe into mode selected above at Marlin startup. Required for 5V mode on 3.0
   * V3.1: Force a probe with unknown mode into selected mode at Marlin startup ( = Probe EEPROM write )
   * To preserve the life of the probe, use this once then turn it off and re-flash.
   */
  //#define BLTOUCH_FORCE_MODE_SET

  /**
   * Use "HIGH SPEED" mode for probing.
   * Danger: Disable if your probe sometimes fails. Only suitable for stable well-adjusted systems.
   * This feature was designed for Delta's with very fast Z moves however higher speed cartesians may function
   * If the machine cannot raise the probe fast enough after a trigger, it may enter a fault state.
   */
  #if NONE(MachineCR10Orig, LowMemoryBoard)
    #define BLTOUCH_HS_MODE
  #endif

  // Safety: Enable voltage mode settings in the LCD menu.
  //#define BLTOUCH_LCD_VOLTAGE_MENU

#endif // BLTOUCH

cchamilt, 2020-02-12:
Yep, thanks.  I'll try
#define BLTOUCH_DELAY 500
if that doesn't work, I'll move on to try changing -
#define BLTOUCH_FORCE_SW_MODE
or
#define BLTOUCH_HS_MODE

jesusbackflips, 2020-02-14:
I had #define BLTOUCH_DELAY 700-900ms
And maybe set 
#define Z_PROBE_LOW_POINT          -1   (max. which Z can move negative below 0 Point)

cchamilt, 2020-02-20:
I did try increasing the delay, but it wasn't any better.
undefining BLTOUCH_HS_MODE seems to be much better so far, but I'll keep testing.

cchamilt, 2020-02-20:
Still really flaky.  Mostly failing ABL, but I can eventually get one done.

It also went into weird control loop issues in ABL.  The BLTouch will fail, then the system will acknowledge the bed level fail.  It will panic/kill the bed and nozzle temp, return to home and then restart the ABL attempt.  It keeps doing that over and over (about 5 times) until it throws a full machine panic and halts the whole system.  

I can try BLTOUCH_FORCE_SW_MODE, but otherwise I am running out of config changes to try.

jesusbackflips, 2020-02-21:
i changed only 2 values...... Delay 600ms and SW mode ON. maybe ur probe, cable are damaged? in SW takes maybe 2-3 sec longer but never failed

InsanityAutomation, 2020-04-04:
Code was changed upstream to prevent the z raise from continuing to the top. For the CRX hs mode was disabled as it was struggling to raise the gantry fast enough. If issues still persist, please reopen the issue.

closed 2020-04-04 by InsanityAutomation

3lack5ky, 2020-10-07:
Hello :)
I‘ve a CR-10s Pro, with a 3.1 BLTouch, which I installed, for the firmware I installed 10sPro_BLT_BIL_DW7.hex. The display is also updated.
I‘be got the same issue as cchamilt, homing is absolut normal, but when the printer tries to measure the bed, the probe goes up and down a couple of times and then resets. It homes again, tries to measure the bed and resets again.

cchamilt, 2020-10-08:
I would recommend trying a replacement.  I had no issues with a new BLTouch on the same wiring, so I have to assume the original was defective.  Way too late for me to return after messing with it for months sadly.

3lack5ky, 2020-10-08:
I tried a known good Sensor from the printer from a friend (BLTouch 2.0). Unfortunately I’ve got the same problems with this one too. :/