InsanityAutomation/Marlin #102: Manual X and Y travel limited to 310mm

mx07423, 2020-02-13:
### Description

in the move screen I cannot make the extruder travel past 310mm on my ender 5 plus in either the X or the Y axis.  The printer limits me when using the X and y arrow keys and if I key in a value greater than 310 it will just ignore the command.  

### Steps to Reproduce

1. Go to the Move screen on the Touch screen display
2. Click on the X or Y arrow Key until to travel to 350mm it will not move that far stops at 310mm
3. Repeat this using the key in value for X or Y axis

**Expected behavior:**  I would expect the X and Y to travel upto my machine limits such for the 5+ it would be 360mm set by the bed size.

**Actual behavior:** Stops at 310mm
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Desktop.zip](https://github.com/InsanityAutomation/Marlin/files/4199042/Desktop.zip)

jesusbackflips, 2020-02-13:
Check this Values here..... 

#define Z_SAFE_HOMING_X_POINT 180
#define Z_SAFE_HOMING_Y_POINT 180

(sourc: 5+ homes to wrong location https://github.com/InsanityAutomation/Marlin/issues/96 ) 
Because in this Firmware Bedsize is set to 360 not 350. It was way off for Homing with 175

InsanityAutomation, 2020-04-04:
This was an issue in the screen files that was resolved with the latest revision.

closed 2020-04-04 by InsanityAutomation