InsanityAutomation/Marlin #105: During auto leveling bed and nozzle heat up

larcin78, 2020-02-29:
### Description
While using the CrealityDwin_2.0 on my CR-10 Max while manual leveling via the lcd the bed and nozzle heat up. The bed to 55 and the nozzle to 215. Not sure this is intended. I can see heating up the be but not the nozzle. 

I found some code in Marlin/Marlin/Configuration_adv.h that is controlling this. 

#if ENABLED(BedDC)
  #define CommBedTmp "55"
#else
  #define CommBedTmp "75"
#endif
  #define USER_DESC_1 "Setup"
  #if (ENABLED(ABL_UBL))
    #define USER_GCODE_1 "M190S" CommBedTmp "\nG28\nG29P1\nG29P3\nG29S1\nG29S0\nG29F0.0\nG29A\nM104S215\nG28\nM109S215\nG1X150Y150F5000\nG1Z0\nM500\nM400\nM117 Set Z Offset"
  #elif ENABLED(ABL_BI)
    #define USER_GCODE_1 "M190S" CommBedTmp "\nG28\nG29\nM400\nM104S215\nG28\nM109S215\nM420S1\nG1X100Y100F5000\nG1Z0\nM500\nM117 Set Z Offset"
  #endif

jkdesignz, 2020-03-16:
#define CommBedTmp "55"  CHANGE 55 TO 10 OR 0
Change the M104S215 TO M104S10 OR 0 
Change M109S215  TO M109S10 OR 0 
Depending on if your using ABL UBL or ABL BI,  change them both to make sure ,

#if ENABLED(BedDC)
#define CommBedTmp "10"
#else
#define CommBedTmp "10"
#endif
#define USER_DESC_1 "Setup"
#if (ENABLED(ABL_UBL))
#define USER_GCODE_1 "M190S" CommBedTmp "\nG28\nG29P1\nG29P3\nG29S1\nG29S0\nG29F0.0\nG29A\nM104S10\nG28\nM109S10\nG1X150Y150F5000\nG1Z0\nM500\nM400\nM117 Set Z Offset"
#elif ENABLED(ABL_BI)
#define USER_GCODE_1 "M190S" CommBedTmp "\nG28\nG29\nM400\nM104S10\nG28\nM109S10\nM420S1\nG1X100Y100F5000\nG1Z0\nM500\nM117 Set Z Offset"
#endif

InsanityAutomation, 2020-03-26:
This is intended as the mesh will be quite different at temperature and most were running it cold them complaining about the results.

closed 2020-03-26 by InsanityAutomation