InsanityAutomation/Marlin #107: SKR PRO: Compile errors 'DGUS_SERIAL' was not declared in this scope

l0l0l0, 2020-03-09:
**Description**
I have a Cr10S PRO / stock screen / SKR PRO 1.1. 
With the new version, impossible to compile, I have the message:

`Compiling .pio\build\BIGTREE_SKR_PRO\src\src\lcd\extensible_ui\lib\dgus\DGUSDisplay.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\lcd\extensible_ui\lib\dgus\DGUSDisplayDefinitionFYSETC.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\lcd\extensible_ui\lib\dgus\DGUSDisplayDefinitionHIPRECY.cpp.o
In file included from Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:1:0:
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In function 'void ExtUI::onStartup()':
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: error: 'DGUS_SERIAL' was not declared in this scope
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:48:2: note: in expansion of macro 'DWIN_SERIAL'
  DWIN_SERIAL.begin(115200);
  ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: note: suggested alternative: 'DWIN_SERIAL'
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:48:2: note: in expansion of macro 'DWIN_SERIAL'
  DWIN_SERIAL.begin(115200);
  ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In member function 'int ExtUI::RTSSHOW::RTS_RecData()':
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: error: 'DGUS_SERIAL' was not declared in this scope
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:363:9: note: in expansion of macro 'DWIN_SERIAL'
  while (DWIN_SERIAL.available() > 0 && (recnum < SizeofDatabuf))
         ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: note: suggested alternative: 'DWIN_SERIAL'
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:363:9: note: in expansion of macro 'DWIN_SERIAL'
  while (DWIN_SERIAL.available() > 0 && (recnum < SizeofDatabuf))
         ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In member function 'void ExtUI::RTSSHOW::RTS_SndData()':
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: error: 'DGUS_SERIAL' was not declared in this scope
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:464:4: note: in expansion of macro 'DWIN_SERIAL'
    DWIN_SERIAL.write(databuf[i]);
    ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: note: suggested alternative: 'DWIN_SERIAL'
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:464:4: note: in expansion of macro 'DWIN_SERIAL'
    DWIN_SERIAL.write(databuf[i]);
    ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In member function 'void ExtUI::RTSSHOW::RTS_SndData(const char*, long unsigned int, unsigned char)':
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: error: 'DGUS_SERIAL' was not declared in this scope
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:500:4: note: in expansion of macro 'DWIN_SERIAL'
    DWIN_SERIAL.write(databuf[i]);
    ^~~~~~~~~~~
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.h:114:23: note: suggested alternative: 'DWIN_SERIAL'
   #define DWIN_SERIAL DGUS_SERIAL
                       ^
Marlin\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:500:4: note: in expansion of macro 'DWIN_SERIAL'
    DWIN_SERIAL.write(databuf[i]);`

Expected behavior: Compile without errors

Actual behavior: Throw compile errors

**Additional Information**

Branche : CrealityDwin_2.0
there is a name error on Line 443: #elif ENABLED (SKR11PRO) instead of #elif ENABLED (SKRPRO11) but the compilation error remains

antoinelb1, 2020-03-17:
Did you succeed building your it to your SKR PRO V1.1 ?
Thanks for the error on line 443. 

Im actually trying to get it to work on my Ender 5 Plus, stock screen ( i believe its a very similar touchscreen as your CR10S Pro) and still have many errors when i try to run it.

mark2382, 2020-04-01:
I was able to get a compile by changing line 114 of Creality_DWIN.h 
from:
 #define DWIN_SERIAL DGUS_SERIAL
to:
#define DWIN_SERIAL Serial1

Pretty ugly, obviously this should be set in Configuration.h. My guess is that all the DGUS_SERIAL_PORT references in Configuration.h starting at line 449-455 should be DGUS_SERIAL with values Serial1, Serial0, Serial2, etc.

InsanityAutomation, 2020-04-04:
Modified to look at MCU target so it may work for other boards i dont explicitly support.

closed 2020-04-04 by InsanityAutomation