InsanityAutomation/Marlin #108: DWIN 2.0 CR10S PRO V1, SCREEN LOCKS AFTER PRINT IS DONE

jkdesignz, 2020-03-15:
Not sure what is wrong, I have to restart my printer after a print is done. everything works great before a print starts, and as soon as the print is done the screen buttons do not work, only printer info and I can go to settings, but movement, homing.  nothings works.  I have to power cycle printer, and when it comes back on, it asks me do I want to resume last print, even though it was done. so I say no. and everything works as is supposed to until I print something else. then I have to do it all over again.

InsanityAutomation, 2020-03-26:
A change was made upstream that in certain conditions could cause an M0 wait for 1 minute looking like a lockup. A bump change is being pushed to resolve this now.

InsanityAutomation, 2020-04-04:
Closing this as the issue looks to be resolved here

closed 2020-04-04 by InsanityAutomation