InsanityAutomation/Marlin #109: BedPID disable, enable Switching(fails, fixed)

jesusbackflips, 2020-03-17:
New Version doesnt uncomment some Lines for BedPid if enabled Bed Switching.

Lines i Uncommenten for easy fix (Plz help me out for functional fix ;) ) THX
( "Marlin-CrealityDwin_2.0/Marlin/src/lcd/extensible_ui/lib/Creality_DWIN.cpp" )

-------LINE 294------
      #if HAS_PID_HEATING
        rtscheck.RTS_SndData(pid_hotendAutoTemp, HotendPID_AutoTmp);
        //rtscheck.RTS_SndData(pid_bedAutoTemp, BedPID_AutoTmp);
        rtscheck.RTS_SndData((unsigned int)(getPIDValues_Kp(E0) * 10), HotendPID_P);
        rtscheck.RTS_SndData((unsigned int)(getPIDValues_Ki(E0) * 10), HotendPID_I);
        rtscheck.RTS_SndData((unsigned int)(getPIDValues_Kd(E0) * 10), HotendPID_D);
        //rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kp() * 10), BedPID_P);
        //rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Ki() * 10), BedPID_I);
        //rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kd() * 10), BedPID_D);
      #endif
-------LINE 303------

-------LINE 1034-----
      #if HAS_PID_HEATING
        else if (recdat.addr == HotendPID_AutoTmp)
          pid_hotendAutoTemp = (uint16_t)recdat.data[0];
        //else if (recdat.addr == BedPID_AutoTmp)
        //  pid_bedAutoTemp = (uint16_t)recdat.data[0];
      #endif
-------LINE 1039-----

-------LINE 1074-----
        #if HAS_PID_HEATING
          else if (recdat.addr == HotendPID_P) {
            setPIDValues(tmp_float_handling*10, getPIDValues_Ki(getActiveTool()), getPIDValues_Kd(getActiveTool()), getActiveTool());
          }
          else if (recdat.addr == HotendPID_I) {
            setPIDValues(getPIDValues_Kp(getActiveTool()), tmp_float_handling*10, getPIDValues_Kd(getActiveTool()), getActiveTool());
          }
          else if (recdat.addr == HotendPID_D) {
            setPIDValues(getPIDValues_Kp(getActiveTool()), getPIDValues_Ki(getActiveTool()), tmp_float_handling*10, getActiveTool());
          }
          //else if (recdat.addr == BedPID_P) {
          //  setBedPIDValues(tmp_float_handling*10, getBedPIDValues_Ki(), getBedPIDValues_Kd());
          //}
          //else if (recdat.addr == BedPID_I) {
          //  setBedPIDValues(getBedPIDValues_Kp(), tmp_float_handling*10, getBedPIDValues_Kd());
          //}
          //else if (recdat.addr == BedPID_D) {
          //  setBedPIDValues(getBedPIDValues_Kp(), getBedPIDValues_Ki(), tmp_float_handling*10);
          //}
        #endif
-------LINE 1093-----

-------LINE 1483-----
//#if HAS_PID_HEATING
        //  case 5: {
        //    SERIAL_ECHOLN("Bed PID");
        //    startBedPIDTune(pid_bedAutoTemp);
        //    break;
        //  }
        //#endif
-------LINE 1489-----

-------LINE 1908-----
#if HAS_PID_HEATING
  void OnPidTuning(const result_t rst) {
    // Called for temperature PID tuning result
    rtscheck.RTS_SndData(pid_hotendAutoTemp, HotendPID_AutoTmp);
        //rtscheck.RTS_SndData(pid_bedAutoTemp, BedPID_AutoTmp);
        rtscheck.RTS_SndData((unsigned int)(getPIDValues_Kp(E0) * 10), HotendPID_P);
        rtscheck.RTS_SndData((unsigned int)(getPIDValues_Ki(E0) * 10), HotendPID_I);
        rtscheck.RTS_SndData((unsigned int)(getPIDValues_Kd(E0) * 10), HotendPID_D);
        //rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kp() * 10), BedPID_P);
        //rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Ki() * 10), BedPID_I);
        //rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kd() * 10), BedPID_D);
  }
#endif
-------LINE 1920-----

-------ERROR MESSAGE------ (not full report)
AppData\Local\Temp\arduino_build_836138\sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In function 'void ExtUI::onIdle()':

AppData\Local\Temp\arduino_build_836138\sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:300:45: error: 'getBedPIDValues_Kp' was not declared in this scope

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kp() * 10), BedPID_P);
-------ERROR MESSAGE------

Greetings < 3

jgagnon5541, 2020-03-21:
I have a similar problem with compile failure on my  ender 5 plus 
only options enabled are:
#define MachineEnder5Plus
#define PLUS
Compile log:
Arduino: 1.8.12 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In function 'void ExtUI::onIdle()':

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:300:45: error: 'getBedPIDValues_Kp' was not declared in this scope

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kp() * 10), BedPID_P);

                                             ^~~~~~~~~~~~~~~~~~

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:300:45: note: suggested alternative: 'getPIDValues_Kp'

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kp() * 10), BedPID_P);

                                             ^~~~~~~~~~~~~~~~~~

                                             getPIDValues_Kp

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:301:45: error: 'getBedPIDValues_Ki' was not declared in this scope

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Ki() * 10), BedPID_I);

                                             ^~~~~~~~~~~~~~~~~~

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:301:45: note: suggested alternative: 'getPIDValues_Ki'

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Ki() * 10), BedPID_I);

                                             ^~~~~~~~~~~~~~~~~~

                                             getPIDValues_Ki

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:302:45: error: 'getBedPIDValues_Kd' was not declared in this scope

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kd() * 10), BedPID_D);

                                             ^~~~~~~~~~~~~~~~~~

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:302:45: note: suggested alternative: 'getPIDValues_Kd'

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kd() * 10), BedPID_D);

                                             ^~~~~~~~~~~~~~~~~~

                                             getPIDValues_Kd

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In member function 'void ExtUI::RTSSHOW::RTS_SndData(int, long unsigned int, unsigned char)':

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:525:26: warning: right shift count >= width of type [-Wshift-count-overflow]

    snddat.data[0] = n >> 16;

                          ^~

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp: In member function 'void ExtUI::RTSSHOW::RTS_HandleData()':

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:1085:52: error: 'getBedPIDValues_Ki' was not declared in this scope

             setBedPIDValues(tmp_float_handling*10, getBedPIDValues_Ki(), getBedPIDValues_Kd());

                                                    ^~~~~~~~~~~~~~~~~~

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:1085:52: note: suggested alternative: 'getPIDValues_Ki'

             setBedPIDValues(tmp_float_handling*10, getBedPIDValues_Ki(), getBedPIDValues_Kd());
[... 152 lines cut ...]

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:1917:45: note: suggested alternative: 'getPIDValues_Ki'

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Ki() * 10), BedPID_I);

                                             ^~~~~~~~~~~~~~~~~~

                                             getPIDValues_Ki

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:1918:45: error: 'getBedPIDValues_Kd' was not declared in this scope

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kd() * 10), BedPID_D);

                                             ^~~~~~~~~~~~~~~~~~

sketch\src\lcd\extensible_ui\lib\Creality_DWIN.cpp:1918:45: note: suggested alternative: 'getPIDValues_Kd'

         rtscheck.RTS_SndData((unsigned int)(getBedPIDValues_Kd() * 10), BedPID_D);

                                             ^~~~~~~~~~~~~~~~~~

                                             getPIDValues_Kd

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

jgagnon5541, 2020-03-21:
Found the problem, define your bed type at line 73 of configuration.h apparently it's not set automatically when defining the type of printer before that.

InsanityAutomation, 2020-03-26:
Ill add changes in the next version that will disable bed pid functions to the screen if the bed is either off or not pid controlled.

closed 2020-03-26 by InsanityAutomation