InsanityAutomation/Marlin #112: Which parameters do I have to set for my stock Ender 5 Plus?

manwald, 2020-03-28:
Hi guys,

I want to use the awesome firmware from TinyMachines, that I already run on my CR10s Pro, also on my Ender 5 Plus.

I see that I have to compile it for myself for the E5P, that's not a big problem for me (I'm using VSCode with Platform.io for this).

But I'm stuck with setting the right parameters in the configuration.h :-(

jesusbackflips, 2020-03-28:
#define MachineEnder5Plus
//#define PLUS >>> careful you also enable ALL METAL HOTEND (allows Temp. up to 300°)!!

#define HotendStock
#define CREALITY_ABL_MOUNT or #define ABL_BLTOUCH

#define ABL_BI
//#define CrealitySilentBoard

#define Z_AFTER_HOMING  5  (mm) Height after homing Z   >> Not scratch Bed after Homing
#define XY_PROBE_SPEED 3000 Speed for Meshprobing, Homing (stock 6000 (mm/s) think)

manwald, 2020-03-28:
Thank you very much!!!

So that's all I have to set/define before compiling the firmware?

One other problem I have is that when I try to compile the FW using VSCode and Platform.io I get this error:
"Error: Currently, PlatformIO does not support debugging for `Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)`."

What's wrong there?

InsanityAutomation, 2020-04-04:
If you are using a standard machine, rather than setting anything in configuration.h, see the predefined environments in VSCode and PlatformIO as they should cover what you need.

closed 2020-04-04 by InsanityAutomation

InsanityAutomation, 2020-04-04:
#define HotendStock
#define ABL_BLTOUCH
These are redundant with plus set above.

#define GraphicLCD //Full graphics LCD for Ender 4, CR-X or CR10SPro
This would be for if you replaced the stock screen with a standard 12864 display

#define AddonFilSensor //Adds a filamnt runout sensor to the CR20 or Ender 4
Stock for this machine so redundant

#define OrigLA // Enable this for linear advance instead of mesh leveling on a melzi board
Only for melzi boards so irrelevant here

manwald, 2020-04-09:
Thanks for your replies!

How do I set the predefined envs in VSCode with PlatformIO?

And do you have any hint how to get rid of the above error-message while trying to compile?