InsanityAutomation/Marlin #113: Which parameters do I have to set for my CR-X (CRX_BLT_UBL_Fil)

LMS0815, 2020-03-30:
All,

I do not have any success compiling a FW for my Creality CR-X (and I cyn not find any examples inf 
`Configurations-bugfix-2.0.x.zip`

This is what I am trying...
Active defines in configuration.h:
```
  #define MachineCRX
  #define HotendStock
  #define BedDC
  #define ABL_BLTOUCH
  #define GraphicLCD
  #define ForceCRXDisplay
  #define ABL_UBL
  #define DualFilSensors
  #define MeshFine

  #define LowMemoryBoard  //  no //#define BLTOUCH_HS_MODE

```
  Active defines in Configuration_adv.h

```
  - http://marlinfw.org/docs/gcode/M851.html
  - Same as UBL Z-Offset
  #define NOZZLE_TO_PROBE_OFFSET { -10.9, -41.6, -2.40 }

```  Compiling for board:
`  Arduino Mega or Mega 2560 / Atmega2560
`

InsanityAutomation, 2020-04-04:
#define GraphicLCD
  #define ForceCRXDisplay

These conflict with each other, are you using a 12864 display or stock display?

LMS0815, 2020-04-05:
My CR-X is having a stock display. 

https://plastikjunkies.de/wp-content/uploads/2018/06/touch.png

Its workinf fine with: CRX_BLT_UBL_Fil_DW6.hex

LMS0815, 2020-04-05:
THX I compiled using
```
//#define GraphicLCD
#define ForceCRXDisplay

```
*.hex file was created. But I get am _low memory_ warning. The file is much bigger than CRX_BLT_UBL_NoFil_DW6.hex.

closed 2020-05-08 by LMS0815